node-usb/node-usb #383: Loop starvation

mildsunrise, 2020-09-10:
If USB events are arriving fast enough, this module can completely starve the Node.js loop, preventing everything else from running.

When poll is used: this seems like a libusb backend-specific bug, [reported here](https://github.com/libusb/libusb/issues/780).
When poll is not used: it happens because there's an [infinite loop](https://github.com/tessel/node-usb/blob/a69f1e9cb83d11d88cb009013b7346f8c3eac3a4/src/uv_async_queue.h#L59) when popping items from the queue & calling callbacks.

erikkallen, 2020-11-20:
I seem to be experiencing the same problem, reading from an fpga, not using polling electron seems to freeze at some point

mildsunrise, 2020-11-22:
@erikkallen you can try to defer event handling to a `setImmediate` call. this way your callback will be short (just the `setImmediate`) and the work will happen later, on its own tick.

but you should also take some other action, or the setImmediate queue will grow bigger and bigger if events arrive faster than you process them

mildsunrise, 2022-03-24:
Update: libusb did fix the loop starvation issues in https://github.com/libusb/libusb/commit/006ca0fbaa447cac329579d1072304833208358c, which landed in v1.0.24.
I haven't tested yet, but upgrading to v1.0.24 should suffice to get fixes on linux / darwin.

thegecko, 2022-03-24:
Thanks, I have tried upgrading to 1.0.24 and had issues. I'll try again...

thegecko, 2022-03-24:
Wow, didn't expect that! It worked first time:

https://github.com/node-usb/node-usb/pull/490

mildsunrise, 2022-03-24:
Nice!! I'm also seeing you're now using NAPI threadsafe functions instead of a custom queue, so that should take care of the loop starvation for Windows. (And if it doesn't it should be Node.js responsibility, not yours)

So when #490 merges, this can basically be closed. I'd like to test someday, let's see if I have time.

thegecko, 2022-03-25:
v1.0.25 is now the libusb version used in usb@v2.2.0