node-usb/node-usb #386: Smart card reader causes libusb crash

xdimh, 2020-10-12:
Insert and unplug the smart card reader on windows platform causes libusb crash.
releated issues [#285](https://github.com/tessel/node-usb/issues/285)

`static char *parse_device_interface_path(const char *interface_path)
{
	char *device_id, *guid_start;
	unsigned int i, len;

	len = safe_strlen(interface_path);
	if (len < 4) {
		return NULL;
	}

	// Microsoft indiscriminatly uses '\\?\', '\\.\', '##?#" or "##.#" for root prefixes.
	if (((interface_path[0] == '\\') && (interface_path[1] == '\\') && (interface_path[3] == '\\')) ||
		((interface_path[0] == '#') && (interface_path[1] == '#') && (interface_path[3] == '#'))) {
		interface_path += 4;
		len -= 4;
	}

	guid_start = strchr(interface_path, '{');
	if (guid_start != NULL) {
		len = (guid_start - interface_path) - 1; // One more for separator
	}

	if (len <= 0) {
		usbi_err(NULL, "program assertion failed: invalid device interface path");
		return NULL;
	}

	device_id = calloc(len + 1, 1); // One additional for NULL term.
	if (device_id == NULL) {
		return NULL;
	}
	strncat(device_id, interface_path, len);

	// Now convert to uppercase and replace '#' with '\'
	for (i = 0; i < len; i++) {
		device_id[i] = (char)toupper((int)device_id[i]);
		if (device_id[i] == '#')
			device_id[i] = '\\';
	}

	return device_id;
}`

The unsigned int len cause the len 
Value cannot be negative
and then Cause the code `if(len<=0)` not to be executed
when len should be -1;

device_id[i] emit Array out of bounds exception

Just need to modify `unsigned int len` to `int len`

danielmain, 2021-09-15:
Having the same Issue. Thank you @xdimh now can I patch it in my project.

Dont you @xdimh want to create a PR so that the node-usb Team can apply it faster?

mildsunrise, 2022-03-25:
IIUC we gave up on using a patched libusb (#453), so v1.9+ currently has no real hotplug support on windows and falls back to periodic polling from JS.

That will hopefully change with #488.

thegecko, 2022-03-26:
Although the current polling mechanism on Windows is sub-standard, I'd still be keen to know if the switch to more recent upstream libusb has fixed this crash (any usb release since November 2021 / usb@1.9.0).

mildsunrise, 2022-03-26:
the crash is from a patch that we used to apply to libusb in order to get hotplug support for windows. the code is no longer there since we switched to upstream libusb

thegecko, 2022-03-26:
Excellent,  closing.  Please re open this if required.

closed 2022-03-26 by thegecko