node-usb/node-usb #388: Too many errors to build on Windows

quinton-ashley, 2020-10-19:
On Windows there are too many errors in one of the files for the project to build.
```
C:\Users\quint\AppData\Local\node-gyp\Cache\10.1.3\include\node\v8.h(9918):
fatal error C1003: error count exceeds 100; stopping compilation
(compiling source file ..\src\device.cc) 
gyp ERR! build error
gyp ERR! stack Error: `C:\Program Files (x86)\MSBuild\14.0\bin\MSBuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (C:\Users\quint\AppData\Roaming\npm\node_modules\npm\node_modules\node-gyp\lib\build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:314:20)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:275:12)
gyp ERR! System Windows_NT 10.0.18362
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\quint\\AppData\\Roaming\\npm\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd E:\dev\apps\nostlan\node_modules\usb
gyp ERR! node -v v12.19.0
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! usb@1.6.3 install: `prebuild-install --verbose || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the usb@1.6.3 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\quint\AppData\Roaming\npm-cache\_logs\2020-10-19T20_20_37_576Z-debug.log
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! nostlan@1.17.3 rebuild: `npm rebuild --build-from-source --runtime=electron --target=10.1.3 --disturl=https://atom.io/download/atom-shell`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the nostlan@1.17.3 rebuild script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\quint\AppData\Roaming\npm-cache\_logs\2020-10-19T20_20_37_595Z-debug.log
```

ngvcanh, 2020-10-20:
What is the terminal that you using?

quinton-ashley, 2020-10-20:
I'm using git bash

quinton-ashley, 2020-10-20:
Same thing happened on cmd as with git bash. I can post the full log if necessary. There are just a lot of warnings and syntax errors.  Too many for MSBuild to let it build.

ngvcanh, 2020-10-20:
Do you have using `--save` option?

```
npm install usb --save
```

quinton-ashley, 2020-10-20:
Yes I've already done that. I have the latest version of npm.  The problem is too many build errors when I try to build it for electron v10.

quinton-ashley, 2020-10-20:
Here's the full log.
[2020-10-20T06_11_20_370Z-debug.log](https://github.com/tessel/node-usb/files/5406633/2020-10-20T06_11_20_370Z-debug.log)

quinton-ashley, 2020-10-20:
Building usb for electron v10 works on macOS and Linux but not on Windows.

quinton-ashley, 2020-10-20:
This project must be updated to VS 2019, it's currently on VS 2015.
[image]

ngvcanh, 2020-10-20:
Here is what I am using:
- Windows 10 Pro x64 - 2004 - 01/09/2020 - 19041.572 - Pack 120.2212.31.0
- NodeJS: `v12.18.3`
- npm: `6.14.8`
- electron: `^10.1.3`
- usb: `^1.6.3`
- prebuild-install: `5.3.5`
- Python: 2.7.15
- windows-build-tools: `5.2.2`

And before writing this comment, I tried creating a new project and installing `usb` package, it still works normally with the following commands:

```
npm install electron electron-builder electron-rebuild --save-dev
npm install usb --save
```

I remember reading somewhere that `usb` package only works with python 2.7. But I'm not sure with that.

Please try to install python 2.7 on your computer and then try again. You also need to install the python 2.7 executable path into the `Path`  system environment

quinton-ashley, 2020-10-20:
What do you mean by it gets replaced by the nostlan package? This same command works on macOS and linux.

quinton-ashley, 2020-10-20:
I have electron electron-builder and electron-rebuild installed globally. The `usb` package needs to start supporting Python 3 or make prebuilt binaries available.

ngvcanh, 2020-10-20:
I have read a bit of the documentation of the `nostlan` package on npmjs.com

It is used as a building tool for the game. And I build the `usb` package on my machine and the tool used is the `node-gyp` package.

I'm also not sure but there might be some conflict inside the `nostlan` package.
You can try.

quinton-ashley, 2020-10-20:
That's an very old version of Nostlan. You can find the current version on my github page.

quinton-ashley, 2020-10-20:
Oh I haven't updated my windows-build-tools that might be the problem!

quinton-ashley, 2020-10-20:
It really works for you? @ngvcanh It's still not working at all, same problem.

ngvcanh, 2020-10-20:
Yes, I have also just read the documentation on the `node-gyp` repository and I plan to come back here asking if you have installed or updated `windows-build-tools`.

I am new to usb package and I am also having some other problems related to usb package.
If you don't mind, can you give me a contact to discuss?
My Skype: ngvcanh.ken
Telegram: Ken Nguyen (+84 979714412)

ngvcanh, 2020-10-20:
I installed on 2 pc and 1 laptop, they are Windows 10.
It normally work.

quinton-ashley, 2020-10-20:
Building for electron v10 isn't not working for you either though or you got it to work?

ngvcanh, 2020-10-20:
I'm using electron v10.
And I installed 2 hours ago

quinton-ashley, 2020-10-20:
In the meantime I will keep using electron v9 on windows since building for it works.

kevinmehall, 2021-01-30:
v1.6.4 has been released with prebuilds for Node 14 and Electron 9, 10, 11, and 12-beta1. Unfortunately I can't individually debug everyone's broken compiler setup. Check out what GitHub Actions does as it builds fine there.

closed 2021-01-30 by kevinmehall