node-usb/node-usb #390: crashing javascript adapter of iobroker when using it via node-co2-monitor

realwax, 2020-10-22:
dmesg:
38.413305] usb 1-1.1.2: usbfs: process 639 (io.javascript.2) did not claim interface 0 before use

`
host.ioBrokerPIB32 | 2020-10-22 15:29:30.779 | error | instance system.adapter.javascript.2 terminated with code 1 (JS_CONTROLLER_STOPPED)
-- | -- | -- | --
host.ioBrokerPIB32 | 2020-10-22 15:29:30.779 | error | Caught by controller[0]: at Transfer.transferDone (/opt/iobroker/node_modules/iobroker.javascript/node_modules/usb/usb.js:454:10)
host.ioBrokerPIB32 | 2020-10-22 15:29:30.779 | error | Caught by controller[0]: at InEndpoint.EventEmitter.emit (domain.js:482:12)
host.ioBrokerPIB32 | 2020-10-22 15:29:30.778 | error | Caught by controller[0]: at InEndpoint.emit (events.js:310:20)
host.ioBrokerPIB32 | 2020-10-22 15:29:30.778 | error | Caught by controller[0]: at Object.onceWrapper (events.js:416:28)
host.ioBrokerPIB32 | 2020-10-22 15:29:30.777 | error | Caught by controller[0]: at InEndpoint.<anonymous> (/opt/iobroker/node_modules/iobroker.javascript/node_modules/node-co2-monitor/co2_monitor.js:90:33)
host.ioBrokerPIB32 | 2020-10-22 15:29:30.770 | error | Caught by controller[0]: at Interface.attachKernelDriver (/opt/iobroker/node_modules/iobroker.javascript/node_modules/usb/usb.js:330:21)
host.ioBrokerPIB32 | 2020-10-22 15:29:30.769 | error | Caught by controller[0]: Error: LIBUSB_ERROR_BUSY
javascript.2 | 2020-10-22 15:29:30.050 | error | at Transfer.transferDone (/opt/iobroker/node_modules/iobroker.javascript/node_modules/usb/usb.js:454:10)
javascript.2 | 2020-10-22 15:29:30.050 | error | at InEndpoint.EventEmitter.emit (domain.js:482:12)
javascript.2 | 2020-10-22 15:29:30.050 | error | at InEndpoint.emit (events.js:310:20)
javascript.2 | 2020-10-22 15:29:30.050 | error | at Object.onceWrapper (events.js:416:28)
javascript.2 | 2020-10-22 15:29:30.050 | error | at InEndpoint.<anonymous> (/opt/iobroker/node_modules/iobroker.javascript/node_modules/node-co2-monitor/co2_monitor.js:90:33)
javascript.2 | 2020-10-22 15:29:30.050 | error | at Interface.attachKernelDriver (/opt/iobroker/node_modules/iobroker.javascript/node_modules/usb/usb.js:330:21)
javascript.2 | 2020-10-22 15:29:30.050 | error | (6318) Error: LIBUSB_ERROR_BUSY
javascript.2 | 2020-10-22 15:29:30.049 | error | (6318) Error: LIBUSB_ERROR_BUSY
javascript.2 | 2020-10-22 15:29:30.047 | error | (6318) An error happened which is most likely from one of your scripts, but the originating script could not be detected.
javascript.2 | 2020-10-22 15:29:30.031 | error | at Transfer.transferDone (/opt/iobroker/node_modules/iobroker.javascript/node_modules/usb/usb.js:441:9)
javascript.2 | 2020-10-22 15:29:30.031 | error | at InEndpoint.EventEmitter.emit (domain.js:482:12)
javascript.2 | 2020-10-22 15:29:30.031 | error | at InEndpoint.emit (events.js:310:20)
javascript.2 | 2020-10-22 15:29:30.031 | error | at InEndpoint.<anonymous> (/opt/iobroker/node_modules/iobroker.javascript/no
`

realwax, 2020-10-27:
There is an issue with raspberry pie's as it seems the kernel has a driver running. The issue originates out of node-co2-monitor using node-usb. As I am using node-co2-monitor and I am not a dev I can't do much about it. Can you please check if claiming usb interfaces works on raspberry pies? cheers.

host.ioBrokerPIB32 | 2020-10-27 14:43:18.283 | error | instance system.adapter.javascript.2 terminated with code 1 (JS_CONTROLLER_STOPPED)
-- | -- | -- | --
host.ioBrokerPIB32 | 2020-10-27 14:43:18.283 | error | Caught by controller[0]: at Transfer.transferDone (/opt/iobroker/node_modules/iobroker.javascript/node_modules/usb/usb.js:454:10)
host.ioBrokerPIB32 | 2020-10-27 14:43:18.282 | error | Caught by controller[0]: at InEndpoint.EventEmitter.emit (domain.js:483:12)
host.ioBrokerPIB32 | 2020-10-27 14:43:18.282 | error | Caught by controller[0]: at InEndpoint.emit (events.js:314:20)
host.ioBrokerPIB32 | 2020-10-27 14:43:18.281 | error | Caught by controller[0]: at Object.onceWrapper (events.js:420:28)
host.ioBrokerPIB32 | 2020-10-27 14:43:18.281 | error | Caught by controller[0]: at InEndpoint.<anonymous> (/opt/iobroker/node_modules/iobroker.javascript/node_modules/node-co2-monitor/co2_monitor.js:90:33)
host.ioBrokerPIB32 | 2020-10-27 14:43:18.280 | error | Caught by controller[0]: at Interface.attachKernelDriver (/opt/iobroker/node_modules/iobroker.javascript/node_modules/usb/usb.js:330:21)
host.ioBrokerPIB32 | 2020-10-27 14:43:18.278 | error | Caught by controller[0]: Error: LIBUSB_ERROR_BUSY

realwax, 2020-10-29:
I had a look at node-co2-monitor and it detached kernel driver before usage even though I get errors above. Can somebody help me further debubbing this as it keeps crashing my javascript adapter in iobroker. Thank you

Apollon77, 2020-11-11:
@kevinmehall I had a look into this and I might not underdstand the code structure by 100% but in fact an Interface has endpoints and in the In/OutENdpoints there could be error s emitted ... but I do not see any place on the interface that "Bubbled up" these error events, so they are thrown instead in an uncactachable mannort ... or !?

Should not Interface also be an Eventemitter and pass error events through evento the device or such?

closed 2023-02-26 by thegecko