node-usb/node-usb #401: Add more maintainers

kevinmehall, 2020-12-20:
While I've been merging small PRs and occasionally making releases, I'm not really using Node much these days, and so my motivation to work on this library has declined.

The project has reached a point where several long-standing issues have become pressing, while I'm not currently able to put in the effort to deal with them. I'm glad to see the project so widely used, but I hate to see the issues and pull requests piling up. I'm hoping there are contributors interested in taking on a more active role.

### Prebuild ABI treadmill

As a C++ extension using V8 / NAN directly, node-usb currently  needs to be built for each new Node and Electron release. Users tend not to have compiler toolchains on Windows, so rely on prebuilds that have to be kept updated.

N-API  (implemented in https://github.com/tessel/node-usb/pull/399) is the way out of this.

### Old libusb

This library is linked against a very outdated fork of libusb, containing my rebase of an old patch from the libusb mailing list adding hotplug event support for Windows. Upstream libusb has still not added hotplug support for Windows, but we're missing out on several years of upstream bugfixes.

I don't know what the best answer here is. Libusb is similarly undermaintained, and I don't expect them to add Windows hotplug support anytime soon. It's possible that node-usb 2.0 should just make the breaking change of not guaranteeing hotplug support on all platforms.

### Testing

The [full test suite](https://github.com/tessel/node-usb/blob/master/test/usb.coffee), which I run prior to releases, requires hardware that was never produced in quantity, running firmware that I seem to have lost the source code for. The firmware is pretty simple, just a couple of endpoints and control requests that echo data back, and it would be good to migrate this to a piece of widely available hardware. (https://github.com/tessel/node-usb/issues/5)

Travis-CI is sunsetting their support for open source projects, necessitating a move to another CI platform, probably GitHub Actions.

### API issues

The API was designed before `Promise` and `async/await` and could use a redesign. Other major deficiencies include [broken handling of USB configurations](https://github.com/tessel/node-usb/issues/381) because that was added as an afterthought. The standard for USB in JS nowadays is WebUSB, so a hypothetical `node-usb` 2.0 should probably just adopt that API wholesale. @thegecko's [webusb](https://github.com/thegecko/webusb) wraps this library in a webusb-compatible API and could be a good starting point.

thegecko, 2020-12-21:
@kevinmehall I'm willing to help out here in some capacity. I have JavaScript/TypeScript skills, but only a little C/C++ knowledge (and even less coffeescript). I would be happy to work on a GitHub actions CI and v2 API.

Regarding the CI, it would probably need to be a well maintained automated test suite as well as physical testing, so potentially a multi-pronged approach. I have access to a large range of Arm Micro controllers, so if a volunteer could help recreate your echo firmware I'd be happy to create a suite (perhaps jest with TypeScript?)

On the API side, looking at the WebUSB standard is very sensible. I can donate some time to that on the JavaScript side and it may even be worth considering considering merging with the `webusb` project (or vice-versa) at some point.

One thing which may be really useful is if a current charter or guidelines could be shared. Even if this is a brain-dump of the existing release processes and architecture thinking (e.g. handling blocking calls).

N-API and libusb work is something where I'd be less effective :(

geovie, 2021-01-07:
@kevinmehall Maybe [abandonware](https://abandonware.github.io/) is interested in taking this over. They use node-usb in [abandonware/node-bluetooth-hci-socket](https://github.com/abandonware/node-bluetooth-hci-socket).

thegecko, 2021-01-12:
> Maybe abandonware is interested in taking this over. They use node-usb in abandonware/node-bluetooth-hci-socket.

Personally I'd rather see it continue here if I can get some support on the C/C++ side

thegecko, 2021-01-12:
> The full test suite, which I run prior to releases, requires hardware that was never produced in quantity

Out of interest, which hardware was used?

I'm going to take a look at seeing how easy it is to get a suite running on more common hardware.

kevinmehall, 2021-01-12:
> Out of interest, which hardware was used?

Atmel atxmega32a4u on [this board](https://github.com/nonolith/Breakouts).

And I think I've found the source code for that firmware: https://kevinmehall.net/tmp/xmega-stream.zip

kevinmehall, 2021-01-12:
@tcr, Could you give @thegecko access to this repository? (I don't seem to have admin permissions myself).

thegecko, 2021-01-26:
> And I think I've found the source code for that firmware

Thanks for that, @kevinmehall . I'm working on porting it to a cheaper/more available device.

I'm seeing one issue with polling when running the test suite multiple times, would you mind confirming on your side that running the test suite multiple times sees a success each time? Would be good to know I'm working with a stable baseline!

kevinmehall, 2021-01-27:
Yes, the test suite successfully runs multiple times without resetting the device. The device does maintain timers, packet counters, and other state, but the tests don't care about the packet contents.

thegecko, 2021-01-30:
Thanks for the source code, it's enabled me to recreate the test hardware (see https://github.com/tessel/node-usb/issues/5#issuecomment-770231673).

thegecko, 2021-02-04:
Another option is to transfer this repo to its own organisation if there is trouble getting admin actions undertaken?

I registered https://github.com/node-usb just in case ;)

kevinmehall, 2021-02-11:
@Frijol, do you still have admin access to this repo? Can you give me and @thegecko admin access?

rwaldron, 2021-02-12:
@kevinmehall I've added @thegecko

kevinmehall, 2021-02-13:
@thegecko I've given you access to publish the package on NPM as well

thegecko, 2021-02-14:
Thanks @kevinmehall First release mad 🤞

thegecko, 2021-04-11:
@rwaldron I'm keen to add some self-hosted runners to GitHub actions on this repo to enable creation of prebuilds not supported by the runners provided by GitHub (e.g. rpi, MacOS M1).

In order to manage these runners, I believe I need to be given admin access to this repo. Is this something you can do?

thegecko, 2021-04-15:
@kevinmehall regarding the libusb version, could you confirm what functionality would be lost on Windows if we move to using the official libusb releases?

Is it the ability to emit events when a known device is disconnected and reconnected?

What are your thoughts on integrating a library such as `libwdi` to avoid users needing to install drivers?

kevinmehall, 2021-04-16:
Correct. The libusb [hotplug APIs](http://libusb.sourceforge.net/api-1.0/libusb_hotplug.html) are unsupported on Windows. This maps to the `'attach'` and `'detach'` events in node-usb. [The version of libusb used in node-usb](https://github.com/kevinmehall/libusb/commits/hp) is based on one of the previous attempts at implementing hotplug on Windows that seemed to work but never got merged upstream.

Libwdi could be useful in some cases but I fear it would be a support burden. With how many people report issues here trying to use HID, CDC, or mass storage devices with this library, I'd want a big warning on it, because someone's going to loop over all their devices and replace all the drivers and break their mouse and keyboard.

The preferred option for custom devices is a [WCID descriptor](https://github.com/pbatard/libwdi/wiki/WCID-Devices) that tells Windows 7+ to bind the WinUSB driver automatically. That's not always an option, but it's a more elegant solution when it is.

thegecko, 2021-05-04:
I'm wondering if switching to `WinUSB` (obviously for Windows only) could be a way to get the features we need. The official `LibUSB` library could then be used for other hosts.

kevinmehall, 2021-06-06:
@rwaldron, can you give @thegecko Admin-level access to this repo so that he can configure GitHub Actions?

thegecko, 2021-10-04:
@rwaldron @tcr @Frijol Bump!

@kevinmehall and myself are requesting admin access to this repository to manage it more effectively, can one of ypu action this, please?

kevinmehall, 2021-10-18:
Thanks, @Frijol! And thanks @thegecko  for taking on maintenance of this library!

closed 2021-10-18 by kevinmehall