node-usb/node-usb #409: Problem with Electron 11

karniv00l, 2021-01-30:
I've just tested the new release (v1.6.4) (https://github.com/tessel/node-usb/pull/407) and it breaks my app on `Electron 11` (tested on macos and ubuntu 20).

However it works on `Electron 10` and `Electron 12 beta 1` 🤔

Error message:

```
Uncaught Error: The module '.../node_modules/usb/build/Release/usb_bindings.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 87. This version of Node.js requires
NODE_MODULE_VERSION 85. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install
```

For reference: https://github.com/nodejs/node/blob/master/doc/abi_version_registry.json

kevinmehall, 2021-01-30:
This is normally what happens if you switch Node / Electron versions without reinstalling the package.

But there is something weird going on here: At https://github.com/tessel/node-usb/releases/tag/v1.6.4 we should have a set of `electron-v87` prebuilds for Electron 12 [as configured](https://github.com/tessel/node-usb/blob/master/.github/workflows/prebuild.yml#L37), but they're weren't generated. So I don't know where you're even getting v87 binaries from unless the files there named v85 actually contain v87 artifacts. That would be a bug in [prebuild](https://github.com/prebuild/prebuild).

Could you `npm install usb` (or electron equivalent) and paste the output lines prefixed with `prebuild-install`? That shows what URL it's downloading.

karniv00l, 2021-01-30:
Thanks for the response. There is also `serialport` in the package.json.

- removed `~/.npm/_prebuilds`
- removed `node_modules` and `package-lock.json `
- `node v14.15.4`

For both Electron `11.0.0` and `12.0.0-beta.1` output is the same 🤔 (`a66dc7-usb-v1.6.4-node-v83-darwin-x64.tar.gz`):

```bash
...
> usb@1.6.4 install /Users/piotrek/git/Ardu-Stim/UI/node_modules/usb
> prebuild-install --verbose || node-gyp rebuild

prebuild-install info begin Prebuild-install version 5.3.6
prebuild-install info looking for cached prebuild @ /Users/piotrek/.npm/_prebuilds/a66dc7-usb-v1.6.4-node-v83-darwin-x64.tar.gz
prebuild-install http request GET https://github.com/tessel/node-usb/releases/download/v1.6.4/usb-v1.6.4-node-v83-darwin-x64.tar.gz
prebuild-install http 200 https://github.com/tessel/node-usb/releases/download/v1.6.4/usb-v1.6.4-node-v83-darwin-x64.tar.gz
...
```

there is also `electron-rebuild` that is running right after npm install:

```bash
> electron-rebuild -f

⠙ Building module: bindings, Completed: 0  CXX(target) Release/obj.target/bindings/src/serialport.o
⠋ Building module: bindings, Completed: 0  CXX(target) Release/obj.target/bindings/src/serialport_unix.o
⠋ Building module: bindings, Completed: 0  CXX(target) Release/obj.target/bindings/src/poller.o
⠼ Building module: bindings, Completed: 0  CXX(target) Release/obj.target/bindings/src/darwin_list.o
⠏ Building module: bindings, Completed: 0  SOLINK_MODULE(target) Release/bindings.node
✔ Rebuild Complete
```

Skipping `electron-rebuild` after npm install actually fixes the problem but throws similar error about `serialport:`

```
node_modules/@serialport/bindings/build/Release/bindings.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 83. This version of Node.js requires
NODE_MODULE_VERSION 85.
```

kevinmehall, 2021-01-30:
`83` is Node 14, so that's getting chosen based on your Node, rather than Electron version. That's a different error than the `85`/`87` mismatch from your original post. I don't really know how `electron-rebuild` is supposed to work, or what's responsible for downloading the right prebuild when using electron.

I confirmed that the `usb-v1.6.4-electron-v85-linux-x64.tar.gz` on the release actually contains a `.node` file build for v87, so even if it were downloading the "right" file, you'd still get the wrong version. I opened https://github.com/prebuild/prebuild/issues/273 for that.

I've had enough of this prebuild ABI madness and am not going to debug this further. Hopefully N-API (#399) will avoid all these problems.

karniv00l, 2021-01-31:
Thanks. That makes more sense. 

`electron-rebuild` produces additional version for `Electron`:

```bash
# ls -la ~/.npm/_prebuilds

20300 Jan 31 01:07 a17f71-bindings-v9.0.4-node-v83-darwin-x64.tar.gz
69405 Jan 31 01:07 a66dc7-usb-v1.6.4-node-v83-darwin-x64.tar.gz
69487 Jan 31 01:08 f6cf1d-usb-v1.6.4-electron-v85-darwin-x64.tar.gz
```

so If the source file (`node-v83`) is invalid, recompiled / rebuilt version (`electron-v85`) will also be invalid because it's based on the wrong source file.

kevinmehall, 2021-01-31:
If it's actually rebuilding from source, the source code isn't tied to any ABI version.

Anyway, I've removed the `usb-v1.6.4-electron-v85-*` prebuild artifacts (that are actually overwritten by ABI 87) from the release, so you should be back to whatever level of brokenness you had before v1.6.4 was released.

thegecko, 2021-01-31:
Argh, this is annoying, looks like `node-abi` is returning the wrong versions, a [similar issue exists for Node 14 too](https://github.com/lgeiger/node-abi/issues/93)

thegecko, 2021-02-14:
Fixed in v1.6.5

closed 2021-02-14 by thegecko