node-usb/node-usb #413: v2.0 API design

thegecko, 2021-02-14:
As discussed in #401, we are considering merging the [webusb](https://github.com/thegecko/webusb) library into `node-usb` in order to take advantage of a standardised USB API - the [WebUSB API specification](https://wicg.github.io/webusb/).

I'd like to consider bringing in and merging the webusb library wholesale, however this comes with some drawbacks:

**Device**
- Potential loss of some of the lower-level usb details such as LIBUSB_* constants, extended descriptors, bus number, port numbers, kernel driver management, etc.
- Potential loss of endpoint polling
- WebUSB device loading would be heavier due to blocking functions needing to be called in order to build the device object. e.g. loading string descriptors.

My recommendation would be to adopt the [WebUSB device](https://wicg.github.io/webusb/#usbdevice) as the shape of all USB device objects and consider unofficial extension points for features not supported above.

**Discovery**
- Potential loss of the simple `getDeviceList()` and `findByIds()` interfaces, instead using the slightly more complicated [requestDevice()](https://wicg.github.io/webusb/#dom-usb-requestdevice).

This could be mitigated by exposing two device selection interfaces - `WebUSB` and `Classic`, both of which return [WebUSB device](https://wicg.github.io/webusb/#usbdevice) objects.

In order to keep the API as simple as possible, I'm keen to understand which features can be deprecated and which must be kept. Can we get away with just using the WebUSB specification, or will this break common use cases (as I suspect).

To help with the new API design, any ideas and thoughts are welcome.

busticated, 2021-05-22:
thank you @thegecko for stepping up to help w/ maintenance 🙏 

one thing i'd love to see considered: bundling all platform binaries into the npm package using [prebuildify](https://github.com/prebuild/prebuildify) as it greatly simplifies consumption in applications with cross-platform support (e.g. CLIs, VSCode extensions, etc). 

fwiw, last i checked, the [node-serialport](https://github.com/serialport/node-serialport) project is [aiming to adopt it](https://github.com/serialport/node-serialport/issues/975#issuecomment-573489942)

koush, 2021-05-30:
I've forked this library and webusb because of the aforementioned issues and here's how I'm approaching it:

Updated to the latest libusb, abandon the fork with hot plug.
Maintaining a libusb fork just compounds maintenance issues. I just want to pull the latest upstream and be done.

The node-usb library should continue to exist as is, maybe even stripped down a bit, as a light verbatim wrapper on libusb. Modernize callbacks with promises and switch to typescript. Remove any  magic (like auto selecting the default configuration which causes my multi configuration devices to have issues). Removed windows hotplug. Removed the JavaScript interning of the libusb device that caused issues with finalization. (Invalid argument error)

The node webusb library continues to function as a layer on top of node-usb. Windows hotplug support can be added using: https://github.com/MadLittleMods/node-usb-detection.
Alternatively, it can target winusb directly as a native module. This is actually what the browser implementation does: chrome bypasses libusb and uses winusb for their webusb implementation now.

koush, 2021-06-01:
Here are my in progress forks:
https://github.com/koush/node-usb
https://github.com/koush/webusb

I've updated to the latest libusb, converted everything to typescript, and promisified the node-usb API.

thegecko, 2021-06-06:
@busticated I think that's worth looking into. As binaries are currently created manually for some targets, I think the challenge will be how to automate usage of prebuildify in CI.

@koush You raise some interesting points. Removal of hotplug support will break backwards compatibility (and personally I've had issues with node-usb-detection on Windows as it doesn't always discover serial numbers correctly). I'm hoping this will be implemented in libusb and plan to switch to the main fork at that point.

We are looking for a solid API to use for this library moving forwards and adopting the webusb API seems the most sensible approach.

FYI, progress towards this can be seen here which includes a lot of the changes already outlined:

https://github.com/thegecko/node-usb/tree/v2

Still very much a work in progress.

koush, 2021-06-06:
That sounds fine, the webusb api is great. I think the low level libusb bindings (this library) should continue to exist because webusb is a limited subset.

incidentally windows hot plug support continued to work even though I switched to upstream libusb which does not have it. Unclear why.

thegecko, 2021-07-10:
I've released a first draft of the v2.0 API for discussion. The source is on the [v2.0.0 branch](https://github.com/tessel/node-usb/tree/v2.0.0) and can be installed from npm using:

```bash
> npm i usb@alpha
```

# Compatibility
This version should be backwards compatible apart from one known breaking type change:

The `controlTransfer` function now returns a `Buffer` **or** `Number` type to support returning the bytes written during an `out` transfer. TypeScript consumers will need to explicitly check the return type for existing programs to continue to work.

# Changes
This version brings some notable changes:

- The JavaScript has been rewritten with strong types using TypeScript split into maintainable classes/files
- TS types have been integrated into the code which means the [DefinitelyTyped](https://github.com/DefinitelyTyped/DefinitelyTyped/tree/master/types/usb) `@types/usb` package is no longer required and types stay aligned with the source
- A simplified [WebUSB](https://wicg.github.io/webusb/) API from [thegecko/webusb](https://github.com/thegecko/webusb) has been included along with some fixes and additional hardware tests
- Source linting with ESLint has been added
- Automated document creation using TypeDoc has been added
- Yarn has been implemented for package management
- There have been no changes to the C source/N-API layer with focus on the JS plumbing

# APIs
This version is a major re-architecture of the JavaScript code and initially I've kept the core `usb` API the same as a starting point for discussion or in case we decide to maintain compatibility.

Therefore, existing code such as the following should still work:

```javascript
import usb from 'usb';
const devices = usb.getDeviceList();
console.log(devices);
```

A default WebUSB API has been added as a property called `webusb` an can be used as follows:

``` javascript
import usb from 'usb';

(async () => {
    // Returns first matching device
    const d = await usb.webusb.requestDevice({
        filters: [{}]
    })
    console.log(d.serialNumber); 
})();
```

To control how devices are selected (e.g. to display a list to the user and mimic the WebUSB popup in the browser), it's also possible to create a new WebUSB class instance specifying a selection function (see https://thegecko.github.io/webusb/docs/interfaces/usboptions.html for more details)

```javascript
import { WebUSB } from 'usb/dist/webusb/index.js';

(async () => {
    const w = new WebUSB({
        devicesFound: (devices => devices.find(device => device.serialNumber === 'TEST_DEVICE'))
    });
    const d = await w.requestDevice({
        filters: [{}]
    });
    console.log(d.serialNumber);
})();
```

Finally, it's possible to merge the two approaches to quickly find a device and then work with it as a WebUSB device. This has been the most common approach I've used when sharing/porting code between the web and desktop:

```javascript
import usb from 'usb';
import { WebUSBDevice } from 'usb/dist/webusb/webusb-device.js';

(async () => {
    const d = await usb.findBySerialNumber('TEST_DEVICE');
    const w = await WebUSBDevice.createInstance(d);
    console.log(w);
})();
```

# Decisions to make
At this point, I'd like to gather some feedback around the next design decisions...

## Do we maintain the existing interface for backwards compatibility?
While it's nice to do this, I'm keen to implement more modern `async/await` style calls and remove the callbacks. This approach would break compatibility, however.

## Do we remove the existing interface entirely?
We could focus on just exposing the WebUSB API, but we would lose quick access to devices (WebUSB device construction is blocking due to strings being read) as well as features such as polling.
It's also difficult to just list all devices using WebUSB as the system needs to replicate the `allowedDevices` functionality to stay inline with the specification.
I tend towards keeping some of the key features in the existing API (e.g. the device listing and finding functions), but allow creation of WebUSB classes from the lightweight usb device once selected (as in the hybrid example above).

## Documentation
Does the typedoc documentation bring any benefit over the readme file? (see the docs/index.html file)

Any other observations/comments around the API design?

thegecko, 2021-07-13:
Regarding the API design, the public-facing parts are now conveniently exposed here:

https://github.com/tessel/node-usb/blob/v2.0.0/tsc/index.ts

This is where I propose to make any major changes and it currently mimics the existing API (plus additions) as outlined in the post above.

My proposal for changes are as follows:

```typescript
import { promisify } from 'util';
import { WebUSB } from './webusb';
import { WebUSBDevice } from './webusb/webusb-device';
import * as usb from './usb';

/**
 * Convenience method to get the first device with the specified VID and PID, or `undefined` if no such device is present.
 * @param vid
 * @param pid
 */
const findByIds = (vid: number, pid: number): usb.Device | undefined => {
    ...
};

/**
 * Convenience method to get the device with the specified serial number, or `undefined` if no such device is present.
 * @param serialNumber
 */
const findBySerialNumber = async (serialNumber: string): Promise<usb.Device | undefined> => {
    ...
};

const webusb = new WebUSB();
const getDeviceList = usb.getDeviceList;

export {
    // Core usb object for quick access
    usb,

    // Convenience methods
    getDeviceList,
    findByIds,
    findBySerialNumber,

    // Default WebUSB object (mimics navigator.usb)
    webusb,

    // WebUSB class for creating custom webusb instances
    WebUSB,

    // WebUSB Device class for turning a core usb.Device into a webusb device
    WebUSBDevice
};

```

cc @kevinmehall

thegecko, 2021-08-30:
I've implemented my proposal above and re-released v2.0.0 for feedback:

```bash
npm i usb@alpha
```

thegecko, 2021-10-14:
@jeremywho The API proposal in this PR addresses the interface design and is tangential to the issue you describe.

I don't believe this PR claims to fix the windows sleep issue which is apparent in the libusb implementation and not the JS wrapper.

I'd love to see this issue fixed and believe it has been in various places. if @stberger3040 or @tombertrand could create a PR on https://github.com/node-usb/libusb we could get a fix tested.

cc @kevinmehall

thegecko, 2022-03-20:
API now implemented

closed 2022-03-20 by thegecko