node-usb/node-usb #416: Build errors on installing this dependency

KevinHernandez8, 2021-04-13:
I have a problem installing this dependency in my bitbucket pipeline. It seems it aims to a different url when it is building files.
I know you made a release 3 days ago with different assets but the installation uses a diferent file and it does not find it.
It is aiming specifically to this URL but I do not why = https://github.com/tessel/node-usb/releases/download/v1.7.0/usb-v1.7.0-napi-v7.1.13-linux-x64.tar.gz

    warning Error running install script for optional dependency: "/usr/local/share/.cache/yarn/v6/.tmp/4687347722768e2d2219861121344c34.529ed7f12a5f21c0b14a8673350f846247fa09c9.prepare/node_modules/usb: Command failed.
    Exit code: 1
    Command: prebuild-install --runtime napi --verbose || node-gyp rebuild
    Arguments: 
    Directory: /usr/local/share/.cache/yarn/v6/.tmp/4687347722768e2d2219861121344c34.529ed7f12a5f21c0b14a8673350f846247fa09c9.prepare/node_modules/usb
    Output:
    prebuild-install info begin Prebuild-install version 5.3.6
    prebuild-install WARN This package does not support N-API version 7.1.13 
    prebuild-install info npm cache directory missing, creating it... 
    prebuild-install info looking for cached prebuild @ /root/.npm/_prebuilds/8994f7-usb-v1.7.0-napi-v7.1.13-linux-x64.tar.gz
    prebuild-install http request GET https://github.com/tessel/node-usb/releases/download/v1.7.0/usb-v1.7.0-napi-v7.1.13-linux-x64.tar.gz
    prebuild-install http 404 https://github.com/tessel/node-usb/releases/download/v1.7.0/usb-v1.7.0-napi-v7.1.13-linux-x64.tar.gz
    prebuild-install WARN install No prebuilt binaries found (target=7.1.13 runtime=napi arch=x64 libc= platform=linux)

Nikola92, 2021-04-14:
I have the same problem today when i'm trying to build electron on windows. Until friday this problem did not show up.

`prebuild-install http request GET https://github.com/tessel/node-usb/releases/download/v1.7.0/usb-v1.7.0-napi-v4.2.12-win32-ia32.tar.gz
    prebuild-install http 404 https://github.com/tessel/node-usb/releases/download/v1.7.0/usb-v1.7.0-napi-v4.2.12-win32-ia32.tar.gz
    prebuild-install WARN install No prebuilt binaries found (target=4.2.12 runtime=napi arch=ia32 libc= platform=win32)
    gyp: Undefined variable napi_build_version in binding.gyp while trying to load binding.gyp
    gyp ERR! configure error
    gyp ERR! stack Error: `gyp` failed with exit code: 1
    gyp ERR! stack     at ChildProcess.onCpExit (C:\Users\Admin\AppData\Roaming\npm\node_modules\npm\node_modules\node-gyp\lib\configure.js:351:16)
    gyp ERR! stack     at ChildProcess.emit (events.js:198:13)
    gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:248:12)
    gyp ERR! System Windows_NT 10.0.19041
    gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\Admin\\AppData\\Roaming\\npm\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
    gyp ERR! cwd C:\newWorkspaceCashleyPlus\cashleyplusapp\dist\electron\UnPackaged\node_modules\usb
    gyp ERR! node -v v10.16.0
    gyp ERR! node-gyp -v v5.0.5
    gyp ERR! not ok
    npm ERR! code ELIFECYCLE
    npm ERR! errno 1
    npm ERR! usb@1.7.0 install: `prebuild-install --runtime napi --verbose || node-gyp rebuild`
    npm ERR! Exit status 1
    npm ERR!
    npm ERR! Failed at the usb@1.7.0 install script.`

pedronieto84, 2021-04-20:
Hi guys, exactly same problem I have here. It's aiming at a Release that doesn't exist.

Nikola92, 2021-04-20:
Hi guys, I managed to solve my problem by uninstalling node version 10.16.0 and then installing node version 10.15.3. That solved my problem when I try to build electron on windows. I get some warnings in the building process for some deprecated modules, but the process finished successfully. Hope this would be helpful

thegecko, 2021-05-05:
Hi all, thanks for the feedback. I'm really struggling to reproduce this. I've tried different versions of Node on Linux and Mac machines and they all find the correct prebuilt binary.

The URLs prebuild is trying are definitely wrong. They shouldn't be specified as an exact version (and certainly not v7!)

https://github.com/tessel/node-usb/releases/download/v1.7.0/usb-v1.7.0-napi-v7.1.13-linux-x64.tar.gz

should be:

https://github.com/tessel/node-usb/releases/download/v1.7.0/usb-v1.7.0-napi-v4-linux-x64.tar.gz

and

https://github.com/tessel/node-usb/releases/download/v1.7.0/usb-v1.7.0-napi-v4.2.12-win32-ia32.tar.gz

should be:

https://github.com/tessel/node-usb/releases/download/v1.7.0/usb-v1.7.0-napi-v4-win32-ia32.tar.gz

One thing you could try is to add `--target 4` to the install script in the `node-usb` `package.json` so it reads:

```json
{
    ...
    "install": "prebuild-install --runtime napi --target 4 --verbose || node-gyp rebuild"
    ...
}
```

Please let me know if this works.

Otherwise, it would be really useful if someone could repro it in a docker environment. Something along these lines:

```bash
docker pull node:10.16.0
docker run --rm -it node:10.16.0 bash
npm install usb@1.7.0
```

thegecko, 2021-05-05:
@KevinHernandez8 @Nikola92 @pedronieto84 I've created a testing release which outlines the change to the install script mentioned above.

Would any of you confirm if this fixes the issue when trying:

```
npm i usb@testing
```

thegecko, 2021-08-30:
I think this has been resolved, please re-open if not.

closed 2021-08-30 by thegecko