node-usb/node-usb #417: 【Help】/lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.28' not found

EvanXzj, 2021-04-20:
## Error Info
When  run `require('usb')` inside `node:10` docker image , I got this error:

```
Error: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.28' not found (required by /node_modules/usb/build/Release/usb_bindings.node)
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:807:18)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)
    at Module.require (internal/modules/cjs/loader.js:692:17)
    at require (internal/modules/cjs/helpers.js:25:18)
    at bindings (/node_modules/bindings/bindings.js:112:48)
    at Object.<anonymous> (/node_modules/usb/usb.js:1:57)
    at Module._compile (internal/modules/cjs/loader.js:778:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
```

## Steps for error reproduction

```
docker pull node:10

docker run --rm -it node:10 bash

npm install usb@1.7.0

node -e "require('usb')"
```

Looking forward to your reply， Thanks!

robertdrolland, 2021-04-28:
I ran into this issue as well when was doing a clean up of unused modules and the usb 1.6.5 was accidentally upgraded to 1.7.0.  I ended up downgrading back to 1.6.5 and the issue went away (changed the package.json for the project I used this in from ^1.6.5 to ~1.6.5 so that it didn't simply grab the latest 1.x).

thegecko, 2021-05-05:
Hi @EvanXzj Thanks for the feedback and repro steps.

I believe this is to do with the version of glibc in the ubuntu used to create the rebuilds. I'll create a testing version with an older ubuntu to see if it fixes the issue.

thegecko, 2021-05-05:
@EvanXzj , @robertdrolland I've recreated the release using `ubuntu 18.04` and believe this has fixed the issue.

Would you mind testing the pre-release using the repro steps?

```
docker pull node:10

docker run --rm -it node:10 bash

npm install usb@testing

node -e "require('usb')"
```

systemmonkey42, 2021-05-05:
Its working for me.

I did a quick test (npm pull usb@testing) before launching my app, and the GLIBC error was gone!

Thanks.

EvanXzj, 2021-05-06:
@thegecko Thanks!!

when 1.7.1-alpha.1 (testing) version will be to the latest version?

closed 2021-05-06 by EvanXzj

thegecko, 2021-05-06:
> when 1.7.1-alpha.1 (testing) version will be to the latest version?

I plan to make a release once we have confirmation that #416 is fixed.

EvanXzj, 2021-05-07:
@thegecko  Okay, Thanks