node-usb/node-usb #418: Slowdown since upgrade from v1.6.5 to v1.7.0

joelpurra, 2021-04-29:
Compared to `node-usb` https://github.com/tessel/node-usb/releases/tag/v1.6.5 (released 2021-02-14), merely upgrading to https://github.com/tessel/node-usb/releases/tag/v1.7.0 (released 2021-04-10) seems to add a long delay (several seconds) just before the program exits.

Can someone else confirm? Are there any timing tests, or at least test logs with timestamps, available for comparison between the versions?

joelpurra, 2021-04-29:
Did some further testing, and can verify that behavior differs between `node-usb` v1.6.5 and v1.7.0, as well as Node.js v16.0.0 and previous versions. Tested on Ubuntu 20.10 and macOS 10.14 (Mojave).

Timing summary:

| | v12.22.1 | v14.16.1 | v16.0.0 |
| --- | ---: | ---: | ---: |
| v1.6.5 | 50 ms | 50 ms | 50 ms |
| v1.7.0 | 8000 ms | 8000 ms | ∞ |

Created a proof-of-concept repository.

- https://github.com/joelpurra/node-usb-issue-418-delay-at-exit

There might be bugs in `index.js`, since it's boiled-down version of [`uvcc`](https://github.com/joelpurra/uvcc) where the issue was originally reported. That said, this test should hopefully work with generally available hardware since it targets standardized UVC cameras. Tested using a Logitech C920.

thegecko, 2021-05-04:
As mentioned here: https://github.com/tessel/node-usb/pull/399#issuecomment-812550154 this is likely to do with the switch to Node-API (which brings a lot of other benefits) and the `ThreadSafeFunction`. Any help appreciated.

joelpurra, 2021-05-04:
@thegecko: had a look, but didn't get far in terms of fixing the delay. (C/C++ isn't my thing.) Sent pull-requests for some minor issues.

thegecko, 2021-11-07:
> I don't believe it's being caused by ThreadSafeFunction

It totally is!

thegecko, 2021-11-14:
I need to reopen this issue as the fix caused other issues. The slow exit reappears in https://github.com/node-usb/node-usb/pull/456

thegecko, 2021-12-05:
Fixed in #460

closed 2021-12-05 by thegecko

joelpurra, 2021-12-06:
@thegecko: great, thanks! Just released [`uvcc`](https://github.com/joelpurra/uvcc) v5.0.2, targeting `node-usb` v1.9.2.

PR #455 (2021-11-08): Threadsafe refs
This PR introduces 2 changes:

- Add ref/unref functions for the threadsafe queue and expose the ones used by hotplug events as discussed in #385 
- Immediately unref the device threadsafe functions to ensure apps exit in a timely manner, fixes #418