node-usb/node-usb #426: Unable to install 1.7.* on Raspian (Debian for Raspberry Pi)

EpicVoyage, 2021-05-30:
Hello,

I use node-usb for a small project on my Raspberry Pi running Raspian. The 1.6 code worked great. With the advent of 1.7, I am seeing installation problems due to _node-addon-api_ and _node-gyp_.

`$ npm i usb@1.7.1
(node:17802) [DEP0022] DeprecationWarning: os.tmpDir() is deprecated. Use os.tmpdir() instead.
 
> node-addon-api@3.0.2 install /home/pi/temp/node_modules/usb/node_modules/node-addon-api
> node-gyp rebuild

gyp: binding.gyp not found (cwd: /home/pi/temp/node_modules/usb/node_modules/node-addon-api) while trying to load binding.gyp
gyp ERR! configure error 
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (/usr/share/node-gyp/lib/configure.js:304:16)
gyp ERR! stack     at emitTwo (events.js:126:13)
gyp ERR! stack     at ChildProcess.emit (events.js:214:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:198:12)
gyp ERR! System Linux 4.19.66+
gyp ERR! command "/usr/bin/node" "/usr/bin/node-gyp" "rebuild"
gyp ERR! cwd /home/pi/temp/node_modules/usb/node_modules/node-addon-api
gyp ERR! node -v v8.11.1
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok 
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian
 
npm ERR! node-addon-api@3.0.2 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the node-addon-api@3.0.2 install script.
npm ERR! This is most likely a problem with the node-addon-api package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls node-addon-api
npm ERR! There is likely additional logging output above.

npm ERR! System Linux 4.19.66+
npm ERR! command "/usr/bin/node" "/usr/bin/npm" "i" "--save" "usb@1.7.1"
npm ERR! cwd /home/pi/temp
npm ERR! node -v v8.11.1
npm ERR! npm -v 1.4.21
npm ERR! code ELIFECYCLE
npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /home/pi/temp/npm-debug.log
npm ERR! not ok code 0`

The same thing happens for a `git clone` and `npm install`. This happens for npm package installations of usb@1.7.0, usb@1.7.1, and usb@testing. usb@1.6.5 can still be installed via npm, and does install correctly.

I do a lot of the development on MacOS. `npm i usb@1.7.1` works without issue there, but that appears to be because of a prebuilt binary:

`> prebuild-install --runtime napi --target 4 --verbose || node-gyp rebuild

prebuild-install info begin Prebuild-install version 5.3.6
prebuild-install info looking for cached prebuild @ /Users/chris/.npm/_prebuilds/a6bd0a-usb-v1.7.1-napi-v4-darwin-x64.tar.gz
prebuild-install info found cached prebuild 
prebuild-install info unpacking @ /Users/chris/.npm/_prebuilds/a6bd0a-usb-v1.7.1-napi-v4-darwin-x64.tar.gz
prebuild-install info unpack resolved to /Users/chris/hello/node_modules/usb/build/Release/usb_bindings.node
prebuild-install info install Successfully installed prebuilt binary!`

Why are we missing `binding.gyp` for the Raspberry Pi? What is the process to build node-usb?

thegecko, 2021-05-30:
Prebuild images for RPi (`linux-arm` and `linux-arm64`) are available and should be downloaded. See:

https://github.com/tessel/node-usb/releases/tag/v1.7.1

I've just tested this on the latest RPi images (10.8) both 32bit and 64bit using node 10.24 and all was fine (prebuilt binaries were downloaded rather than compiling).

Can you share:

- debian version
- OS architecture
- Node version

EpicVoyage, 2021-05-31:
Thank you for the quick reply! I do see the arm prebuild packages here in github. Here are the system specs:

> $ cat /etc/debian_version
> 9.13
> 
> $ cat /etc/os-release
> PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
> NAME="Raspbian GNU/Linux"
> VERSION_ID="9"
> VERSION="9 (stretch)"
> VERSION_CODENAME=stretch
> ID=raspbian
> ID_LIKE=debian
> HOME_URL="http://www.raspbian.org/"
> SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
> BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
> 
> $ uname -a
> Linux tempsensors 4.19.66+ #1253 Thu Aug 15 11:37:30 BST 2019 armv6l GNU/Linux
> 
> $ node -v
> v8.11.1

thegecko, 2021-05-31:
Node-api requires at least node v10. Upgrading your node version should resolve your issue.

EpicVoyage, 2021-05-31:
Okay, that makes sense. I am using an old Raspberry Pi (v1) with an armv6l processor no longer supported by node. Let me see what I can do about a software upgrade and I will report back.

EpicVoyage, 2021-06-03:
After upgrading from Raspbian Stretch (9) [to Raspbian Buster](https://www.raspberrypi.org/blog/buster-the-new-version-of-raspbian/) (10), I can install node v10 and use node-usb. Thanks for pointing me in the right direction!

closed 2021-06-03 by EpicVoyage