node-usb/node-usb #431: Issues getting Node-USB built Electron

DrGeoffEdwards, 2021-06-24:
I am on windows 10, relatively new to NodeJS.  I have been trying to install Node-USB and have followed many dead ends.  I am using electron-rebuild with the command  ./node_modules/.bin/electron-rebuild and it progresses through much of the process until here :

⠋ Building module: usb, Completed: 0gyp info spawn C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\MSBuild\Current\Bin\MSBuild.exe
gyp info spawn args [
gyp info spawn args   'build/binding.sln',
gyp info spawn args   '/clp:Verbosity=minimal',
gyp info spawn args   '/nologo',
gyp info spawn args   '/p:Configuration=Release;Platform=x64',
gyp info spawn args   map: [Function (anonymous)]
gyp info spawn args ]
⠙ Building module: usb, Completed: 0Building the projects in this solution one at a time. To enable parallel build, please add the "-m" switch.
⠏ Building module: usb, Completed: 0  core.c
⠧ Building module: usb, Completed: 0  descriptor.c
⠹ Building module: usb, Completed: 0  hotplug.c
⠦ Building module: usb, Completed: 0  io.c
⠙ Building module: usb, Completed: 0  strerror.c
⠼ Building module: usb, Completed: 0D:\Users\geoff\code\traci-apptest\node_modules\usb\libusb\libusb\strerror.c(108,3): warning C4566: character represented by universal-character-name '\u0

THEN, it goes into this mode 

⠦ Building module: usb, Completed: 0/union [D:\Users\geoff\code\traci-apptest\node_modules\usb\build\usb_bindings.vcxproj]
D:\Users\geoff\code\traci-apptest\node_modules\usb\libusb\libusb\libusb.h(1259,38): message : This member will be ignored by a defaulted constructor or c 
opy/move assignment operator [D:\Users\geoff\code\traci-apptest\node_modules\usb\build\usb_bindings.vcxproj]
⠧ Building module: usb, Completed: 0D:\Users\geoff\code\traci-apptest\node_modules\node-addon-api\napi.h(1058,24): error C2061: syntax error: identifier 'napi_async_context' [D:\Users\geoff
\code\traci-apptest\node_modules\usb\build\usb_bindings.vcxproj]
D:\Users\geoff\code\traci-apptest\node_modules\node-addon-api\napi.h(1061,24): error C2061: syntax error: identifier 'napi_async_context' [D:\Users\geoff 
\code\traci-apptest\node_modules\usb\build\usb_bindings.vcxproj]

Then finally 

✖ Rebuild Failed

An unhandled error occurred inside electron-rebuild
node-gyp failed to rebuild 'D:\Users\geoff\code\traci-apptest\node_modules\usb'.
Error: `C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\MSBuild\Current\Bin\MSBuild.exe` failed with exit code: 1

Error: node-gyp failed to rebuild 'D:\Users\geoff\code\traci-apptest\node_modules\usb'.
Error: `C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\MSBuild\Current\Bin\MSBuild.exe` failed with exit code: 1

    at ModuleRebuilder.rebuildNodeGypModule (D:\Users\geoff\code\traci-apptest\node_modules\electron-rebuild\lib\src\module-rebuilder.js:193:19)
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
    at async Rebuilder.rebuildModuleAt (D:\Users\geoff\code\traci-apptest\node_modules\electron-rebuild\lib\src\rebuild.js:190:9)
    at async Rebuilder.rebuild (D:\Users\geoff\code\traci-apptest\node_modules\electron-rebuild\lib\src\rebuild.js:152:17)
    at async D:\Users\geoff\code\traci-apptest\node_modules\electron-rebuild\lib\src\cli.js:146:9

Telling me to "find the solution here"  or "try every possible version of NodeJS" perhaps ain't gonna help because much of the above I do not understand how to follow through on.  Is there someone who can provide a "recipe" for installation?

thegecko, 2021-08-30:
This basic example may help:

https://github.com/node-usb/node-usb-example-electron

thegecko, 2021-11-08:
Closing due to inactivity, please reopen this issue if it persists with the latest release.

closed 2021-11-08 by thegecko