node-usb/node-usb #432: Error: LIBUSB_ERROR_ACCESS

qkren, 2021-06-24:
`E:\BACK\record-node-usb\node_modules\usb\usb.js:284
        this.device.__claimInterface(this.id)
                    ^

Error: LIBUSB_ERROR_ACCESS
    at Interface.claim (E:\BACK\record-node-usb\node_modules\usb\usb.js:284:14)
    at Object.<anonymous> (E:\BACK\record-node-usb\index.js:143:21)
    at Module._compile (internal/modules/cjs/loader.js:778:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)
    at Function.Module.runMain (internal/modules/cjs/loader.js:831:12)
    at startup (internal/bootstrap/node.js:283:19)
    at bootstrapNodeJSCore (internal/bootstrap/node.js:623:3)`

node 10.20.0 64  successed, but node 10.20.0 32  is error.

thegecko, 2021-06-24:
This error is usually thrown if the device is in use (claimed) by another program/ process or the udev rules are incorrect when using Linux

thegecko, 2021-07-10:
Closing due to no more comments

closed 2021-07-10 by thegecko