node-usb/node-usb #438: [Bug]: Using node-usb on the Electron main process by exiting a crash popup is displayed

danielmain, 2021-07-30:
### Electron Version

13.1.7

### What operating system are you using?

macOS

### Operating System Version

macOS 11.5

### What arch are you using?

x64

### Last Known Working Electron version

8.5.5

### Expected Behavior

After exit Electron, no crash is expected

### Actual Behavior

When I close electron on macOS after having created a listener using node-usb and calling removeListener on 'window-all-closed', Electron throws an exception EXC_BAD_ACCESS (SIGSEGV)

`Crashed Thread:        0  CrBrowserMain  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       EXC_I386_GPFLT
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [34544]

Thread 0 Crashed:: CrBrowserMain  Dispatch queue: com.apple.main-thread
0   com.github.Electron.framework 	0x000000010a6f02a0 napi_delete_reference + 48
1   usb_bindings.node             	0x0000000113a8b6c0 0x113a88000 + 14016
2   libsystem_c.dylib             	0x00007fff20400d25 __cxa_finalize_ranges + 316
3   libsystem_c.dylib             	0x00007fff20401010 exit + 53
4   libdyld.dylib                 	0x00007fff204ecf44 start + 8
`

### Testcase Gist URL

https://github.com/danielmain/electron-crash-using-usb

### Additional Information

I have created a stupid simple electron application that reproduce the problem. 

Repo: https://github.com/danielmain/electron-crash-using-usb

1. Clone it
2. Run: yarn install && yarn start
3. Close the application and after some seconds you will see the crash

danielmain, 2021-07-30:
If I disable the line node_usb.cc:175 there is no crash

`// hotplugThis.Reset(info.This().As<Napi::Object>(), 1);`

danielmain, 2021-08-10:
Thanks to @jjeff we found out that the crash by closing the app, happens only BigSur and Monterrey. we could not reproduce it on MacOS Catalina.

jjeff, 2021-08-10:
I am seeing this issue as well on an M1 Macbook Air with Big Sur running the (x64) Electron app under Rosetta.

thegecko, 2021-08-13:
Awesome work on recreating this issue with a repo. Please bear with me checking the fix, I'm on vacation.

thegecko, 2021-08-30:
Fixed with https://github.com/tessel/node-usb/pull/440

closed 2021-08-30 by thegecko