node-usb/node-usb #439: electron-rebuild error

Yongle-Fu, 2021-08-10:
MacOS 11.5.1 (20G80)

electron_rebuild_test/node_modules/node-addon-api/napi.h:4:10: fatal error: 'node_api.h' file not found
#include <node_api.h>
         ^~~~~~~~~~~~
⠏ Building module: usb, Completed: 01 error generated.
make: *** [Release/obj.target/usb_bindings/src/node_usb.o] Error 1
✖ Rebuild Failed

An unhandled error occurred inside electron-rebuild

Yongle-Fu, 2021-08-10:
https://github.com/Yongle-Fu/electron_rebuild_test

danielmain, 2021-08-10:
> git clone git@github.com:Yongle-Fu/electron_rebuild_test.git                                        
Cloning into 'electron_rebuild_test'...
remote: Enumerating objects: 15, done.
remote: Counting objects: 100% (15/15), done.
remote: Compressing objects: 100% (13/13), done.
remote: Total 15 (delta 1), reused 15 (delta 1), pack-reused 0
Receiving objects: 100% (15/15), 5.39 KiB | 5.39 MiB/s, done.
Resolving deltas: 100% (1/1), done.

> cd electron_rebuild_test                                                                           
> yarn install                                                            
yarn install v1.22.10
info No lockfile found.
[1/4] 🔍  Resolving packages...
[2/4] 🚚  Fetching packages...
[3/4] 🔗  Linking dependencies...
[4/4] 🔨  Building fresh packages...
success Saved lockfile.
✨  Done in 81.34s.
> npx electron-rebuild                                     
npm verb cli [
npm verb cli   '/Users/daniel/.nvm/versions/node/v14.17.0/bin/node',
npm verb cli   '/Users/daniel/.nvm/versions/node/v14.17.0/lib/node_modules/npm/bin/npm-cli.js',
npm verb cli   'exec',
npm verb cli   '--',
npm verb cli   'electron-rebuild'
npm verb cli ]
npm info using npm@7.20.2
npm info using node@v14.17.0
npm timing npm:load:whichnode Completed in 0ms
npm timing config:load:defaults Completed in 0ms
npm timing config:load:file:/Users/daniel/.nvm/versions/node/v14.17.0/lib/node_modules/npm/npmrc Completed in 2ms
npm timing config:load:builtin Completed in 3ms
npm timing config:load:cli Completed in 1ms
npm timing config:load:env Completed in 0ms
npm timing config:load:file:/Users/daniel/Workspace/electron_rebuild_test/.npmrc Completed in 1ms
npm timing config:load:project Completed in 2ms
npm timing config:load:file:/Users/daniel/.npmrc Completed in 1ms
npm timing config:load:user Completed in 1ms
npm timing config:load:file:/Users/daniel/.nvm/versions/node/v14.17.0/etc/npmrc Completed in 0ms
npm timing config:load:global Completed in 0ms
npm timing config:load:validate Completed in 1ms
npm timing config:load:credentials Completed in 1ms
npm timing config:load:setEnvs Completed in 0ms
npm timing config:load Completed in 9ms
npm timing npm:load:configload Completed in 10ms
npm timing npm:load:setTitle Completed in 11ms
npm timing npm:load:setupLog Completed in 0ms
npm timing config:load:flatten Completed in 2ms
npm timing npm:load:cleanupLog Completed in 1ms
npm timing npm:load:configScope Completed in 0ms
npm timing npm:load:projectScope Completed in 0ms
npm timing npm:load Completed in 28ms
npm http fetch GET 200 https://registry.npmjs.org/electron-rebuild 124ms (cache updated)
npm timing arborist:ctor Completed in 1ms
npm timing arborist:ctor Completed in 0ms
Need to install the following packages:
  electron-rebuild
⸨⠂⠂⠂⠂⠂⠂⠂⠂⠂⠂⠂⠂⠂⠂⠂⠂⠂⠂⸩ ⠋ : timing arborist:ctor Completed in 0ms
npm timing arborist:ctor Completed in 0ms
npm timing idealTree:init Completed in 3ms
npm timing idealTree:userRequests Completed in 0ms
npm http fetch GET 200 https://registry.npmjs.org/electron-rebuild 12ms (cache hit)
npm http fetch GET 200 https://registry.npmjs.org/lzma-native 82ms (cache miss)
npm http fetch GET 200 https://registry.npmjs.org/detect-libc 108ms (cache revalidated)
npm http fetch GET 200 https://registry.npmjs.org/fs-extra 112ms (cache revalidated)
npm http fetch GET 200 https://registry.npmjs.org/colors 143ms (cache revalidated)
npm http fetch GET 200 https://registry.npmjs.org/node-abi 142ms (cache revalidated)
npm http fetch GET 200 https://registry.npmjs.org/tar 149ms (cache updated)
npm http fetch GET 200 https://registry.npmjs.org/@malept%2fcross-spawn-promise 155ms (cache miss)
[... 808 lines cut ...]
⠦ Building module: usb, Completed: 0In file included from ../src/transfer.cc:1:
In file included from ../src/node_usb.h:14:
In file included from /Users/daniel/.electron-gyp/13.1.9/include/node/node_buffer.h:25:
In file included from /Users/daniel/.electron-gyp/13.1.9/include/node/node.h:73:
/Users/daniel/.electron-gyp/13.1.9/include/node/node_version.h:99:9: warning: 'NAPI_VERSION' macro redefined [-Wmacro-redefined]
#define NAPI_VERSION  7
        ^
<command line>:15:9: note: previous definition is here
#define NAPI_VERSION 8
        ^
⠸ Building module: usb, Completed: 01 warning generated.
⠼ Building module: usb, Completed: 0  c++ -bundle -framework CoreFoundation -framework IOKit -undefined dynamic_lookup -Wl,-no_pie -Wl,-search_paths_first -mmacosx-version-min=10.7 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk -arch x86_64 -L./Release -stdlib=libc++  -o Release/usb_bindings.node Release/obj.target/usb_bindings/src/node_usb.o Release/obj.target/usb_bindings/src/device.o Release/obj.target/usb_bindings/src/transfer.o Release/usb.a
⠏ Building module: usb, Completed: 0ld: warning: object file (Release/usb.a(io.o)) was built for newer macOS version (10.13) than being linked (10.7)
ld: warning: object file (Release/usb.a(core.o)) was built for newer macOS version (10.13) than being linked (10.7)
ld: warning: object file (Release/usb.a(descriptor.o)) was built for newer macOS version (10.13) than being linked (10.7)
ld: warning: object file (Release/usb.a(hotplug.o)) was built for newer macOS version (10.13) than being linked (10.7)
ld: warning: object file (Release/usb.a(darwin_usb.o)) was built for newer macOS version (10.13) than being linked (10.7)
ld: warning: object file (Release/usb.a(sync.o)) was built for newer macOS version (10.13) than being linked (10.7)
ld: warning: object file (Release/usb.a(threads_posix.o)) was built for newer macOS version (10.13) than being linked (10.7)
ld: warning: object file (Release/usb.a(poll_posix.o)) was built for newer macOS version (10.13) than being linked (10.7)
✔ Rebuild Complete
npm timing command:exec Completed in 217992ms
npm verb exit 0
npm timing npm Completed in 218787ms
npm info ok`

danielmain, 2021-08-10:
No problems using your repo

thegecko, 2021-08-10:
Thanks @danielmain. This looks like an environment issue, so closing...

closed 2021-08-10 by thegecko