node-usb/node-usb #441: Occasionally an error occurs when calling findByIds / usb.getDeviceList()

zbs-ha, 2021-08-16:
### Environment
- win7 64bit
- electron 12
- usb 1.7.1

### Error stack
[image]

### What went wrong
```js
exports.findByIds = function(vid, pid) {
	var devices = usb.getDeviceList()   // ** That's what went wrong here

	for (var i = 0; i < devices.length; i++) {
		var deviceDesc = devices[i].deviceDescriptor
		if ((deviceDesc.idVendor == vid) && (deviceDesc.idProduct == pid)) {
			return devices[i]
		}
	}
}
```

zbs-ha, 2021-08-16:
When I called again it was normal again, Does anyone know why, thanks!

20051231, 2021-08-20:
I have same problem with electron 13.
It looks like related with V8 version. The exception is thrown at node_usb.cc line 125
`
arr.Set(i, Device::get(env, devs[i]));
`

The Set function will throw error when the Device object is somehow garbage collected?
When I put all devices enumerated by getDeviceList into a Set to prevent them from garbage collected, subsequent getDeviceList will not trigger the error.

thegecko, 2021-08-30:
Can a reliable reproducer for this issue be created?

20051231, 2021-08-30:
I think it only happen in Electron renderer process preload script.

package.json
```
{
  "dependencies": {
    "electron": "^13.2.3",
    "usb": "^1.7.1"
  }
}
```

test.js
```

const usb = require('usb');

async function run() {
    const list = await usb.getDeviceList();
    console.log(list.map(it => `${it.deviceDescriptor.idVendor}:${it.deviceDescriptor.idProduct}`));
    await usb.getDeviceList();
}

async function loop() {
    while(true) {
        await run();
        await new Promise((resolve) => setTimeout(resolve, 1000));
    }
}

loop().catch(e => {
    console.error(e);
})
```

main.js
```
const { BrowserWindow, app } = require('electron');
const { join } = require('path')
// require('./test') // seems working properly with Electron main process

app.on('ready', () => {
    const window = new BrowserWindow({
        webPreferences: {
            preload: join(__dirname, "test.js"),
            // nodeIntegration: true,
            // contextIsolation: false,
        }
    })
    window.loadURL("about:blank");
    // also working fine with renderer process
    // window.loadURL(join(__dirname, "index.html"));
    window.webContents.openDevTools();
})
```

index.html
```
<!DOCTYPE html>
<html>

</html>

<script>
    require('./test.js')
</script>
```

I had played around different way to load the module in Electron. When loaded in main process or web context in renderer process, it works fine. But when it runs in preload script, it will crash almost every second time it calls `getDeviceList` on my machine.

Workaround: just save all devices it gathered and the bug will not happen.

thegecko, 2021-08-30:
Thanks @20051231 

I'm glad you have found a workaround here. It looks like it may be a race condition within LIBUSB.
Any help resolving this would be appreciated.

gniezen, 2022-01-05:
This is consistently happening for me in Electron v16. Not quite sure I understand the workaround?

Anything I can do to provide more details for debugging this?

20051231, 2022-01-06:
@gniezen my workaround is whenever you get an instance of device, store it in a map to prevent it from garbage collection.

```
const set = new Set(); // put this in global scope

const list = await usb.getDeviceList();
for(let device of list)
  set.add(list)
```

but if any device is unplugged, the crash still may happens. In my case, from 50% to 10%. I am considering to move my logic into main process instead of renderer process.

mildsunrise, 2022-03-24:
I'm consistently hitting this, on regular time intervals (once every ~minutes) in Node.js 14.
I'll try to debug.

mildsunrise, 2022-03-24:
Okay, it took me a while to realize this "Invalid argument" error message does *not* come from libusb, it seems to be an N-API error: https://github.com/nodejs/node/blob/718be08686312d19645861b9715491a49ee3c9cb/src/js_native_api_v8.cc#L730
which is thrown as a C++ exception by the N-API C++ wrapper.

maybe `napi_invalid_arg` is thrown [here](https://github.com/node-usb/node-usb/blob/007e702c3053f4cf649356b4cb3994595a82885e/src/node_usb.cc#L142), when attempting the conversion to `Device`? in that case it should be easy to fix

gniezen, 2022-03-25:
Amazing work, thanks @mildsunrise !

mildsunrise, 2022-03-25:
I left the application running overnight with #492 applied and hit no errors.
So I think this can be closed when #492 is merged.

closed 2022-03-25 by thegecko

wizzer129, 2024-03-21:
> I left the application running overnight with #492 applied and hit no errors. So I think this can be closed when #492 is merged.

We have an old forked node-usb branch, recently encountered this issue upgrading node versions. Applied your changes in devices.cc, they worked! Thank you so much

commit 0b463ff7 (2022-03-25): handle devices that are already GC'd (fixes #441) (#492)