node-usb/node-usb #442: How to get user visible drive name (ex. D:)

Bellisario, 2021-08-28:
I need to know the device location accessible by user when he uses an external usb drive, for example, I need only detect that user have connected usb drives on a specific drive (['D:', 'E:']). Is this possible with this library? I don't know how to get an array, as the previous example.

kevinmehall, 2021-08-29:
This library only does low-level USB transfers at the interface / endpoint level for custom devices without kernel drivers. Drive letters happen at a totally different layer (kernel's built-in support for Mass Storage Class devices and filesystems), so this is not the right library for that purpose.

closed 2021-08-29 by kevinmehall

Bellisario, 2021-08-29:
Ok, thanks anyway!

thegecko, 2021-08-30:
I made a stab at something similar a while back:

https://www.npmjs.com/package/usb-mountpoint