node-usb/node-usb #444: Prebuild fixes

mildsunrise, 2021-08-30:
Hi! This is a list of problems with current prebuilds, that I may attempt to fix if I have time:

- [ ] Current prebuilds aren't very portable, since (after the release you are making right now) they target a recent-ish glibc (2.27).
   Unlike musl, on glibc you **must** build against a very old glibc in order to get decent backwards compatibility.
   Targeting glibc <=2.23 gives about 3 years of compatibility.
   I personally recommend building on debian jessie, which has glibc 2.19.

- [ ] **Important:** When the prebuild fails to load, fall back to a manual build. Otherwise it's not possible to use the module at all.
    This is recommended practice and is what every module using `node-gyp-build` does.

- [ ] Ship all prebuilds inside the NPM package.
    It is also good practice, for many reasons, to build prebuilds with `prebuildify` and ship them on the NPM package.

- [ ] musl (i.e. Alpine Linux) support.
    This would also be a nice to have, but is reasonable to exclude.

thegecko, 2021-08-30:
Thanks @mildsunrise , any help would be appreciated.

Comments:

## glibc

All prebuilds are done using debian "buster" based systems, but depending on which prebuild binary you are looking at, they come from different places.

Linux 64 uses a hosted runner for GH actions:
I dropped the linux build runner to ubuntu 18.04 (buster) previously to help with this. It seems to be the oldest supported runner with GH actions since 16.04 (stretch) was [deprecated](https://github.blog/changelog/2021-04-29-github-actions-ubuntu-16-04-lts-virtual-environment-will-be-removed-on-september-20-2021/)
In order to use an older runner, it would need to be self-hosted.

Linux 32 is created locally on an old 32 bit machine running ubuntu 18.04 (buster)

The RPi/Arm Linux builds are created locally using raspbian (buster)

## Fallback
Hmm, this should be working if a prebuilt binary isn't found. Do you see otherwise?

## Prebuildify
I would like to use prebuildify, but currently many of the prebuilds require manual creation locally. I'm concerned layering in prebuildify right now would make the release process quite complicated. I'm awaiting admin access to set up self-hosted runners for these binaries and once that is implemented I think prebuildify could be unblocked.

## Alpine
Yes please!

mildsunrise, 2021-08-30:
> The RPi/Arm Linux builds are created locally using raspbian (buster)

Hmm, I'd personally recommend using docker + qemu, this way you can create prebuilds for all the archs+distros in one go. You first do:

    docker run --rm --privileged multiarch/qemu-user-static --reset -p yes

And then you can `docker run` images from whatever archs you want. It's substantially slower, but much more simple/portable/automated and you have precise control of the image (I use `node:10-jessie`). Also adding musl prebuilds is easy, just i.e. duplicate the prebuilds but with the `node:10-alpine` image.

This is the way I do it in my projects. I just have to run `npm publish`, and `prepublishOnly` creates all the prebuilds without any action from my part. However I understand it will be more complicated here because of the additional OSes.

mildsunrise, 2021-08-30:
> Hmm, this should be working if a prebuilt binary isn't found. Do you see otherwise?

The thing is that if a prebuild binary *is* found, the build is always skipped, without testing if that binary actually works.
So if you are using Alpine, or an old distro or whatever, `npm install usb` will skip the build and the binary will just fail to load when you try to `require("usb")`.

After downloading the binary, we should try to `require()` it to make sure it's valid, and fall back to node-gyp otherwise.

closed 2021-10-18 by thegecko