node-usb/node-usb #457: electron-rebuild with mismatched NAPI_VERSION fails

DuBistKomisch, 2021-11-22:
Using 1.9.1.

```
  g++ -o Release/obj.target/usb_bindings/src/node_usb.o ../src/node_usb.cc '-DNODE_GYP_MODULE_NAME=usb_bindings' '-DUSING_UV_SHARED=1' '-DUSING_V8_SHARED=1' '-DV8_DEPRECATION_WARNINGS=1' '-DV8_DEPRECATION_WARNIN
GS' '-DV8_IMMINENT_DEPRECATION_WARNINGS' '-D_LARGEFILE_SOURCE' '-D_FILE_OFFSET_BITS=64' '-DV8_COMPRESS_POINTERS' '-DV8_31BIT_SMIS_ON_64BIT_ARCH' '-DV8_REVERSE_JSARGS' '-D__STDC_FORMAT_MACROS' '-DOPENSSL_NO_PINSH
ARED' '-DOPENSSL_THREADS' '-DNAPI_VERSION=8' '-DBUILDING_NODE_EXTENSION' -I../src -I/home/jake/_redacted_/node_modules/usb/node_modules/node-addon-api -I/home/jake/.electron-gyp/11.2.3/include/node
 -I/home/jake/.electron-gyp/11.2.3/src -I/home/jake/.electron-gyp/11.2.3/deps/openssl/config -I/home/jake/.electron-gyp/11.2.3/deps/openssl/openssl/include -I/home/jake/.electron-gyp/11.2.3/deps/uv/include -I/ho
me/jake/.electron-gyp/11.2.3/deps/zlib -I/home/jake/.electron-gyp/11.2.3/deps/v8/include -I../libusb/libusb  -fPIC -pthread -Wall -Wextra -Wno-unused-parameter -m64 -O3 -fno-omit-frame-pointer -fno-rtti -std=gnu
++1y -std=c++14 -MMD -MF ./Release/.deps/Release/obj.target/usb_bindings/src/node_usb.o.d.raw   -c
⠏ Building module: usb, Completed: 3In file included from /home/jake/.electron-gyp/11.2.3/include/node/node.h:73,
                 from /home/jake/.electron-gyp/11.2.3/include/node/node_buffer.h:25,
                 from ../src/node_usb.h:14,
                 from ../src/node_usb.cc:1:
/home/jake/.electron-gyp/11.2.3/include/node/node_version.h:99: warning: "NAPI_VERSION" redefined
   99 | #define NAPI_VERSION  6
      |
<command-line>: note: this is the location of the previous definition
```

"command line" seems to mean this define: https://github.com/node-usb/node-usb/blob/1361698cfc62bcd9f09e0a22fa636d5e02982235/binding.gyp#L29

The node version I'm using (14.18.1) has NAPI_VERSION 8, whereas my electron version 11.2.3 uses node 12.18.3 with NAPI_VERSION 6.

This then causes the `node-addon-api` header to try to wrap a bunch of functions which exist in NAPI_VERSION 8 but not NAPI_VERSION 6:

```
In file included from /home/jake/_redacted_/node_modules/usb/node_modules/node-addon-api/napi.h:2905,
                 from ../src/node_usb.h:13,
                 from ../src/node_usb.cc:1:
/home/jake/_redacted_/node_modules/usb/node_modules/node-addon-api/napi-inl.h: In member function ‘Napi::MaybeOrValue<bool> Napi::Object::Freeze()’:
/home/jake/_redacted_/node_modules/usb/node_modules/node-addon-api/napi-inl.h:1535:24: error: ‘napi_object_freeze’ was not declared in this scope; did you mean ‘napi_object_expected’?
 1535 |   napi_status status = napi_object_freeze(_env, _value);
      |                        ^~~~~~~~~~~~~~~~~~
      |                        napi_object_expected
/home/jake/_redacted_/node_modules/usb/node_modules/node-addon-api/napi-inl.h: In member function ‘Napi::MaybeOrValue<bool> Napi::Object::Seal()’:
/home/jake/_redacted_/node_modules/usb/node_modules/node-addon-api/napi-inl.h:1540:24: error: ‘napi_object_seal’ was not declared in this scope; did you mean ‘napi_object’?
 1540 |   napi_status status = napi_object_seal(_env, _value);
      |                        ^~~~~~~~~~~~~~~~
      |                        napi_object
/home/jake/_redacted_/node_modules/usb/node_modules/node-addon-api/napi-inl.h: In member function ‘bool Napi::ArrayBuffer::IsDetached() const’:
/home/jake/_redacted_/node_modules/usb/node_modules/node-addon-api/napi-inl.h:1746:24: error: ‘napi_is_detached_arraybuffer’ was not declared in this scope; did you mean ‘napi_create_arraybuffer’?
 1746 |   napi_status status = napi_is_detached_arraybuffer(_env, _value, &detached);
      |                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                        napi_create_arraybuffer
/home/jake/_redacted_/node_modules/usb/node_modules/node-addon-api/napi-inl.h: In member function ‘void Napi::ArrayBuffer::Detach()’:
/home/jake/_redacted_/node_modules/usb/node_modules/node-addon-api/napi-inl.h:1752:24: error: ‘napi_detach_arraybuffer’ was not declared in this scope; did you mean ‘napi_create_arraybuffer’?
 1752 |   napi_status status = napi_detach_arraybuffer(_env, _value);
      |                        ^~~~~~~~~~~~~~~~~~~~~~~
      |                        napi_create_arraybuffer
```

Am I doing something wrong? Is that `binding.gyp` define really correct?

DuBistKomisch, 2021-11-22:
Looks like for 1.7.x, I only get the warning and no errors because `node-addon-api` was on an older version presumably without the newer NAPI_VERSION support.

thegecko, 2021-11-22:
You shouldn't need to rebuild usb for electron

DuBistKomisch, 2021-11-22:
Yes, I think I can workaround this by explicitly listing packages for `electron-rebuild` to rebuild. This still seems like there's something wrong though, do you think it's actually a bug in `electron-rebuild` instead?

DuBistKomisch, 2021-11-23:
Okay, looks like `electron-rebuild` isn't handling the different `libc` options on Linux. I'll raise a bug there, that should help avoid running into this.

However, I still think you should be able to build this module manually. I get the same errors when using `node-gyp` directly i.e. `node-gyp clean && node-gyp configure --target=11.2.3 --dist-url=https://electronjs.org/headers && node-gyp build`, so I'll raise a bug there as well about `napi_build_version` using the host version instead of target.

DuBistKomisch, 2021-11-23:
Looks like `node-gyp` already fixed this issue in 8.4.0, but it only works with Electron 14.2.0 and above, ugh. See https://github.com/nodejs/node-gyp/issues/2534#issuecomment-976089582.

Definitely not a bug with `node-usb` after all!

closed 2021-11-23 by DuBistKomisch

thegecko, 2021-11-23:
Thanks for the awesome investigation