node-usb/node-usb #458: Using usb crashes Electron app in Raspberry Pi 3(Raspbian)

mhcgolds, 2021-11-29:
Im developing an simple electron app for raspberry for personal use to print tickets with thermal printers. To achieve that, Im using a usb thermal printer device plugged in a Raspberry Pi 3 with escpos package([https://github.com/song940/node-escpos](url)) which uses this package.

The thing is that I have developed the whole app in Windows and it works perfectly, but whein I tried to run on raspberry, it crashes. It tracked down the issue and isolated it to this package. The most simple use of it crashes the app already:

    const usb = require('usb');
    const devices = usb.getDeviceList();

The code above is enought to crash the app. I doesn't shows exception in console or in bash, the app just get blank(the loaded html content disappears) and Dev Tools disconnects.

I've produced a simple test case for this issue in [this repo](https://github.com/mhcgolds/electron-usb-rasp-test), which consists in very basic electron app trying to list usb devices(basicly, the code above). It works on Windows 10, but not in Raspberry Pi 3. Note that I just made the app today, so all modules are up to date. The RPi OS is a Raspbian Desktop installation taken from the official site couple days ago, so its up to date as well.

An important information is that it works in a simple nodejs script on RPi environment. I've tested *usb* module and it listed all available usb devices. Tested *escpos* module and it managed to work and successfully printed a test text. It only won't work in an Electron app.

thegecko, 2021-11-29:
Could you try with the example electron app?

https://github.com/node-usb/node-usb-example-electron

It's not been tested in RPi, but would be good to know if it works.

mhcgolds, 2021-11-29:
Just tried it, but its throwing an SIGSEGV error with a clean installation of your test, running electron with version 16. Downgraded it to version 15.0.0, but the same error. Yesterday a spent like 5~6 hours on this and I had that same problem with my main app, but i did sooo much other stuff about all those problems that I can't remember what solved it, but i solved it. 

The error:

```
pi@raspberrypi:~/Desktop/node-usb-example-electron $ npm start

> node-usb-example-electron@1.0.0 start
> electron .

/home/pi/Desktop/node-usb-example-electron/node_modules/electron/dist/electron exited with signal SIGSEGV
npm ERR! code 1
npm ERR! path /home/pi/Desktop/node-usb-example-electron
npm ERR! command failed
npm ERR! command sh -c electron .

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2021-11-21T00_30_57_511Z-debug.log
```

Now i tried with `sudo`, root user directly, installing electron globally, none worked. Not sure how to make it work and I'm getting kinda tired of it (those are the things i really hate in linux). Even [Electron Quick Start](https://github.com/electron/electron-quick-start) didn't works, but my app still works. So idk whats the problem.

I'm afraid I won't be able to test what you asked.

mhcgolds, 2021-11-30:
Probably related to [this issue](https://github.com/electron/electron/issues/31123).

thegecko, 2022-05-07:
Please check this issue still arises with the latest release; v2.4.0

closed 2023-02-26 by thegecko