node-usb/node-usb #459: getDeviceList hangs node.js

MikeColeGuru, 2021-12-04:
I am not certain if this has been reported before.

I am developing a project using node-usb and it is working great except that node.js will not exit without telling it to with `process.exit(0)` 

I wrote a simple program that hangs.
`
const usb = require("usb");
usb.getDeviceList();
`

I am using 

- node v16.13.0 
- libusb 1.0.23-2build1  (ubuntu package)
- Ubuntu 20.04  

I did notice but I'm not certain it is relevant from [this page](https://libusb.sourceforge.io/api-1.0/group__libusb__dev.html#gac0fe4b65914c5ed036e6cbec61cb0b97)

> You are expected to unreference all the devices when you are done with them, and then free the list with libusb_free_device_list(). Note that libusb_free_device_list() can unref all the devices for you. **Be careful not to unreference a device you are about to open until after you have opened it.** 

It appears that `libusb_free_device_list` is being called at the end of `getDeviceList` and before it is opened.

But I am not certain what is causing the hang or how to work around it.

thegecko, 2021-12-04:
Can you confirm that this is actually a delay? In this case,  we are aware of an issue that the clean up of node-api code causes a 3-5 second delay on exit

MikeColeGuru, 2021-12-04:
I ran it for 3 minutes with no exit.   I can try longer if needed.

thegecko, 2021-12-04:
Excellent testing!
Not seen this issue before

MikeColeGuru, 2021-12-04:
I have some more info..
I started commenting out things in the C code until it freed up the exit.
This is what I found...

If I comment out these lines in `uv_async_queue.h` the above simple example will exit.

```
        void start(Napi::Env env) {
            Napi::Function empty_func = Napi::Function::New(env, [](const Napi::CallbackInfo& cb) {});
            //tsfn = Napi::ThreadSafeFunction::New(env, empty_func, "libusb", 0, 1);
        }

        void stop() {
            //tsfn.Release();
        }
```
Lines 15 and 19

MikeColeGuru, 2021-12-04:
It appears to never destroy the devices

```
~/Projects/node-usb-test$ node index.js 
node-usb [../src/device.cc:Device() 25]: Created device 0x5a54790
node-usb [../src/device.cc:Device() 25]: Created device 0x5a544c0
node-usb [../src/device.cc:Device() 25]: Created device 0x593a460
node-usb [../src/device.cc:Device() 25]: Created device 0x59e71d0
node-usb [../src/device.cc:Device() 25]: Created device 0x5a4c9a0
node-usb [../src/device.cc:Device() 25]: Created device 0x59c5150
node-usb [../src/device.cc:Device() 25]: Created device 0x59abd50
node-usb [../src/device.cc:Device() 25]: Created device 0x5923710
node-usb [../src/device.cc:Device() 25]: Created device 0x59d84a0
node-usb [../src/device.cc:Device() 25]: Created device 0x5923830
node-usb [../src/device.cc:Device() 25]: Created device 0x593cb30
node-usb [../src/device.cc:Device() 25]: Created device 0x5a4fbf0
node-usb [../src/device.cc:Device() 25]: Created device 0x5938550
^C
```

It also appears to free them and exit if I expose GC and call it.

thegecko, 2021-12-04:
Thanks for investigating this.  It may also be the cause of https://github.com/node-usb/node-usb/issues/418
Any help creating a fix would be appreciated.

MikeColeGuru, 2021-12-05:
It looks like I have a fix.  I'm still testing but so far it looks promising.

I moved the start and stop of the completion queue to the open and close methods.

https://github.com/MikeColeGuru/node-usb/commit/8ffbfd15dddd3c367a0d9405b8c1cd3da8c0987a

MikeColeGuru, 2021-12-05:
Also, does there need to be a completion queue for every device?  In an experiment, I moved it out of Device and just passed a pointer to it in each Device::get.  Seems to work.

MikeColeGuru, 2021-12-05:
Here are some links that may help going forward.

**Event Handling**
http://libusbx.sourceforge.net/api-1.0/group__asyncio.html#asyncevent

**Make Node.js exit regardless of a native module async call pending**
https://stackoverflow.com/questions/58241265/make-node-js-exit-regardless-of-a-native-module-async-call-pending

**libusb_context**
https://libusb.sourceforge.io/api-1.0/group__libusb__lib.html#ga4ec088aa7b79c4a9599e39bf36a72833

thegecko, 2021-12-05:
> It looks like I have a fix. I'm still testing but so far it looks promising.
I moved the start and stop of the completion queue to the open and close methods.
MikeColeGuru@8ffbfd1

Excellent work, your patch passes all the hardware tests and does indeed seem to fix this issue and #418 

Could you open a PR, please?

MikeColeGuru, 2021-12-05:
Sure ...  https://github.com/node-usb/node-usb/pull/460

thegecko, 2021-12-05:
Fixed in #460

closed 2021-12-05 by thegecko