node-usb/node-usb #461: Add ability to switch to usbdk backend

andrew-welker, 2021-12-08:
After much reading and trying to get this library and some that rely on it to work with a Logitech C930e camera, I ran across this issue (https://github.com/libusb/libusb/issues/247) in the libusb repo. The suggestion there was to switch the backend that libusb is using to something called usbdk (https://github.com/daynix/UsbDk). There's a snippet of code in the libusb wiki that describes how to (possibly) do this (https://github.com/libusb/libusb/wiki/Windows#Driver_Installation), but I'm not skilled enough with C++ to be able to trust that any changes I make will not break something else. 

I would think that adding a function to change the backend would be useful. I'm also willing to be wrong.

thegecko, 2021-12-08:
This sounds like an interesting addition, can any C++ gurus help out?

andrew-welker, 2021-12-09:
Found a little more information. Based on the documentation for libusb on [SourceForge](https://libusb.sourceforge.io/api-1.0/group__libusb__lib.html#ga07d4ec54cf575d672ba94c72b3c0de7c), the `libusb_set_option` call that's used to set the backend option should be called immediately after the `libusb_init` call occurs. Looking at the code, this occurs in the `Init` method, which, I'm not sure is called from the JS side, which means that the `Init` function may need to be exposed in some way or some other way of setting an option to use the `usbdk` backend needs to be set up so that it can be called in the appropriate place.

andrew-welker, 2021-12-10:
Well...I forked the repo and made the modification myself. Turns out, it won't solve my issue, as usbdk obtains exclusive access to the device, and I really only need access to the control interface of the camera for my application. This means that conferencing applications that should be able to continue sending video can't cause the device disappears for them. I still think it makes sense to add this as an option, but it won't solve my issue.

PR #489 (2022-03-24): Introduce useUsbDkBackend function
Fixes #461
closed 2022-03-24 by thegecko