node-usb/node-usb #463: Webpack binding loaders struggle to parse bindings import

freshollie, 2021-12-16:
*TLDR*: I'm not sure if you are willing or need to make these changes. This write-up might just be so that others who experience this issue understand the fix.

With `1.x`, node bindings were imported in a way which was parseable by webpack bindings parsers like [node-bindings-loader](https://www.npmjs.com/package/node-bindings-loader).

With `2.0.0`, these parsers are not clever enough to resolve the `path` import used to resolve the bindings: https://github.com/node-usb/node-usb/blob/master/tsc/usb/bindings.ts#L9

```bash
ERROR in ./node_modules/usb/dist/usb/bindings.js
Module build failed (from ./node_modules/node-bindings-loader/index.js):
NonErrorEmittedError: (Emitted value instead of an instance of Error) ReferenceError: path_1 is not defined
    at processResult (./node_modules/webpack/lib/NormalModule.js:748:12)
    at ./node_modules/webpack/lib/NormalModule.js:853:5
    at ./node_modules/loader-runner/lib/LoaderRunner.js:399:11
    at ./node_modules/loader-runner/lib/LoaderRunner.js:251:18
    at context.callback ./node_modules/loader-runner/lib/LoaderRunner.js:124:13)
    at Object.run (./node_modules/node-bindings-loader/index.js:142:16)
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
```

It would be more compatible if you didn't use `path.join` here, and just used a template string. I know that this seems like a weird limitation.

```ts
const usb = require('node-gyp-build')(`${__dirname}/../..`);
```

I'm using this patch
```patch
diff --git a/dist/usb/bindings.js b/dist/usb/bindings.js
index a80a550c29a45ff7ab2231c903da3c7ddc23be21..6c05c970ee9c35aa68e752d28ba9156233939dfa 100644
--- a/dist/usb/bindings.js
+++ b/dist/usb/bindings.js
@@ -3,8 +3,7 @@
 //  Eric Brody <https://github.com/underscorebrody>
 //  Rob Moran <https://github.com/thegecko>
 Object.defineProperty(exports, "__esModule", { value: true });
-var path_1 = require("path");
 /* eslint-disable @typescript-eslint/no-var-requires */
-var usb = require('node-gyp-build')(path_1.join(__dirname, '..', '..'));
+var usb = require('node-gyp-build')(__dirname + '/../..');
 module.exports = usb;
 //# sourceMappingURL=bindings.js.map
\ No newline at end of file
```

thegecko, 2021-12-17:
Interesting. The danger with not using path.join is removes insurances the path is built correctly for the target OS.

freshollie, 2021-12-17:
I'll try that, but I'm doubtful. These libraries just try to evaluate whatever is in the brackets after the `node-gyp-build`. I think because normally these imports are considered static and so node-style static string pathing is conventionally used.

RReverser, 2022-02-04:
> Interesting. The danger with not using path.join is it removes insurances the path is built correctly for the target OS.

All systems supported by Node.js are happy to accept `/` as a delimiter, so that shouldn't be a problem.

thegecko, 2022-03-18:
I think you may be able to work around this with the `__dirname` setting in webpack 5:

https://webpack.js.org/api/module-variables/#__dirname-nodejs

thegecko, 2023-02-26:
Closing this issue due to inactivity, please create a new issue as required.

closed 2023-02-26 by thegecko

sl1pkn07, 2023-06-23:
hi. i have the same issue in this project (https://github.com/EdgeTX/buddy) when try to build with electron 22 series

ERROR in ./node_modules/usb/dist/usb/bindings.js
Module build failed (from ./node_modules/node-bindings-loader/index.js):
NonErrorEmittedError: (Emitted value instead of an instance of Error) ReferenceError: path_1 is not defined

webpack is set to version 5 in the package.json ("webpack": "^5.64.4",)

how i can fix this issue? i'm not coder. nodejs is bast new for me. only can edit here and there

greetings