node-usb/node-usb #464: LIBUSB_ERROR_NOT_FOUND

Det0xix, 2021-12-29:
Hi, I'm trying to print with my thermal printer (Netum G5), I'm using node-escpos. The error comes from usb package.

```
const escpos = require('escpos');
escpos.USB = require('escpos-usb');

const usbDevice = new escpos.USB(1155, 23057);
console.log('success')

const options = { encoding: "GB18030" /* default */ }

const device = new escpos.Printer(usbDevice, options);

usbDevice.open(function(error){
  printer
  .font('a')
  .align('ct')
  .style('bu')
  .size(1, 1)
  .text('The quick brown fox jumps over the lazy dog')
  .text('敏捷的棕色狐狸跳过懒狗')
  .barcode('1234567', 'EAN8')
  .table(["One", "Two", "Three"])
  .tableCustom(
    [
      { text:"Left", align:"LEFT", width:0.33, style: 'B' },
      { text:"Center", align:"CENTER", width:0.33},
      { text:"Right", align:"RIGHT", width:0.33 }
    ],
    { encoding: 'cp857', size: [1, 1] } // Optional
  )
  .qrimage('https://github.com/song940/node-escpos', function(err){
    this.cut();
    this.close();
  });
});
```

That's the error I get

```
PS C:\Users\jurek\Desktop\ESCPOS> node app.js
success
C:\Users\jurek\Desktop\ESCPOS\node_modules\usb\dist\usb\device.js:80
        this.__open();
             ^

Error: LIBUSB_ERROR_NOT_FOUND
    at Device.ExtendedDevice.open (C:\Users\jurek\Desktop\ESCPOS\node_modules\usb\dist\usb\device.js:80:14)
    at USB.open (C:\Users\jurek\Desktop\ESCPOS\node_modules\escpos-usb\index.js:110:15)
    at Object.<anonymous> (C:\Users\jurek\Desktop\ESCPOS\app.js:11:11)
    at Module._compile (node:internal/modules/cjs/loader:1097:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1149:10)
    at Module.load (node:internal/modules/cjs/loader:975:32)
    at Function.Module._load (node:internal/modules/cjs/loader:822:12)
    at Function.executeUserEntryPoint [as runMain] (node:internal/modules/run_main:81:12)
    at node:internal/main/run_main_module:17:47 {
  errno: -5
}

Node.js v17.3.0
PS C:\Users\jurek\Desktop\ESCPOS> 
```

I tried installing drivers, but it doesn't work

thegecko, 2023-02-26:
This error is coming from libusb and is often due to the OS not allowing access. Sorry I can't help more than that.

closed 2023-02-26 by thegecko

bablu1921, 2023-09-13:
how to allow access can you please help