node-usb/node-usb #465: Issue with webpack and electron-forge

jp-pino, 2021-12-29:
I've recently upgraded an electron-forge project that was using v1.9.1 to v2.0.3. As usual, when I run `npm run start` it starts preparing native dependencies, which recompiles node-usb. This process runs fine:
```
> file-loader@1.0.0 start
> electron-forge start

✔ Checking your system
✔ Locating Application
⠹ Preparing native dependencies: 0 / 1  CC(target) Release/obj.target/libusb/libusb/libusb/core.o
⠹ Preparing native dependencies: 0 / 1  CC(target) Release/obj.target/libusb/libusb/libusb/descriptor.o
⠦ Preparing native dependencies: 0 / 1  CC(target) Release/obj.target/libusb/libusb/libusb/hotplug.o
⠇ Preparing native dependencies: 0 / 1  CC(target) Release/obj.target/libusb/libusb/libusb/io.o
⠏ Preparing native dependencies: 0 / 1../libusb/libusb/io.c:2300:30: warning: comparison of integers of different signs: 'int' and 'nfds_t' (aka 'unsigned int') [-Wsign-compare]
                for (i = internal_nfds ; i < nfds ; ++i) {
                                         ~ ^ ~~~~
⠙ Preparing native dependencies: 0 / 11 warning generated.
⠹ Preparing native dependencies: 0 / 1  CC(target) Release/obj.target/libusb/libusb/libusb/strerror.o
⠼ Preparing native dependencies: 0 / 1  CC(target) Release/obj.target/libusb/libusb/libusb/sync.o
⠦ Preparing native dependencies: 0 / 1  CC(target) Release/obj.target/libusb/libusb/libusb/os/poll_posix.o
⠧ Preparing native dependencies: 0 / 1  CC(target) Release/obj.target/libusb/libusb/libusb/os/threads_posix.o
⠏ Preparing native dependencies: 0 / 1  CC(target) Release/obj.target/libusb/libusb/libusb/os/darwin_usb.o
⠇ Preparing native dependencies: 0 / 1  LIBTOOL-STATIC Release/usb.a
⠸ Preparing native dependencies: 0 / 1  CXX(target) Release/obj.target/usb_bindings/src/node_usb.o
⠸ Preparing native dependencies: 0 / 1../src/node_usb.cc:118:2: warning: 'libusb_set_debug' is deprecated [-Wdeprecated-declarations]
        libusb_set_debug(usb_context, info[0].As<Napi::Number>().Int32Value());
        ^
../libusb/libusb/libusb.h:1324:1: note: 'libusb_set_debug' has been explicitly marked deprecated here
LIBUSB_DEPRECATED_FOR(libusb_set_option)
^
../libusb/libusb/libusb.h:89:49: note: expanded from macro 'LIBUSB_DEPRECATED_FOR'
#define LIBUSB_DEPRECATED_FOR(f) __attribute__((deprecated))
                                                ^
⠸ Preparing native dependencies: 0 / 11 warning generated.
⠼ Preparing native dependencies: 0 / 1  CXX(target) Release/obj.target/usb_bindings/src/device.o
⠧ Preparing native dependencies: 0 / 1  CXX(target) Release/obj.target/usb_bindings/src/transfer.o
⠸ Preparing native dependencies: 0 / 1  SOLINK_MODULE(target) Release/usb_bindings.node
✔ Preparing native dependencies: 1 / 1
✔ Compiling Main Process Code
✔ Launch Dev Servers
✔ Compiling Preload Scripts
✔ Launching Application
```

I can clearly see that it's producing the build files successfully: 
```
ls -lR node_modules/usb/build/                     
total 176
-rw-r--r--  1 jp.pinob  staff  13806 Dec 29 09:02 Makefile
drwxr-xr-x  7 jp.pinob  staff    224 Dec 29 09:02 Release
-rw-r--r--  1 jp.pinob  staff    120 Dec 29 09:02 binding.Makefile
-rw-r--r--  1 jp.pinob  staff  13194 Dec 29 09:02 config.gypi
-rwxr-xr-x  1 jp.pinob  staff  30293 Dec 29 09:02 gyp-mac-tool
-rw-r--r--  1 jp.pinob  staff    114 Dec 29 09:02 libusb.Makefile
-rw-r--r--  1 jp.pinob  staff   6437 Dec 29 09:02 libusb.target.mk
-rw-r--r--  1 jp.pinob  staff   6619 Dec 29 09:02 usb_bindings.target.mk

node_modules/usb/build//Release:
total 1488
drwxr-xr-x  4 jp.pinob  staff     128 Dec 29 09:02 obj.target
-rw-r--r--  1 jp.pinob  staff  426600 Dec 29 09:02 usb.a
-rwxr-xr-x  1 jp.pinob  staff  328344 Dec 29 09:02 usb_bindings.node

node_modules/usb/build//Release/obj.target:
total 0
drwxr-xr-x  3 jp.pinob  staff  96 Dec 29 09:02 libusb
drwxr-xr-x  3 jp.pinob  staff  96 Dec 29 09:02 usb_bindings

node_modules/usb/build//Release/obj.target/libusb:
total 0
drwxr-xr-x  3 jp.pinob  staff  96 Dec 29 09:02 libusb

node_modules/usb/build//Release/obj.target/libusb/libusb:
total 0
drwxr-xr-x  9 jp.pinob  staff  288 Dec 29 09:02 libusb

node_modules/usb/build//Release/obj.target/libusb/libusb/libusb:
total 488
-rw-r--r--  1 jp.pinob  staff  76264 Dec 29 09:02 core.o
-rw-r--r--  1 jp.pinob  staff  50664 Dec 29 09:02 descriptor.o
-rw-r--r--  1 jp.pinob  staff  19440 Dec 29 09:02 hotplug.o
-rw-r--r--  1 jp.pinob  staff  62824 Dec 29 09:02 io.o
drwxr-xr-x  5 jp.pinob  staff    160 Dec 29 09:02 os
-rw-r--r--  1 jp.pinob  staff   8888 Dec 29 09:02 strerror.o
-rw-r--r--  1 jp.pinob  staff  19040 Dec 29 09:02 sync.o

node_modules/usb/build//Release/obj.target/libusb/libusb/libusb/os:
total 376
-rw-r--r--  1 jp.pinob  staff  176168 Dec 29 09:02 darwin_usb.o
-rw-r--r--  1 jp.pinob  staff    3792 Dec 29 09:02 poll_posix.o
-rw-r--r--  1 jp.pinob  staff    4704 Dec 29 09:02 threads_posix.o

node_modules/usb/build//Release/obj.target/usb_bindings:
total 0
drwxr-xr-x  5 jp.pinob  staff  160 Dec 29 09:02 src

node_modules/usb/build//Release/obj.target/usb_bindings/src:
total 3592
-rw-r--r--  1 jp.pinob  staff  734912 Dec 29 09:02 device.o
-rw-r--r--  1 jp.pinob  staff  538928 Dec 29 09:02 node_usb.o
-rw-r--r--  1 jp.pinob  staff  473592 Dec 29 09:02 transfer.o
```

But they're not being added into the webpack outputs:
```
.webpack:
total 0
drwxr-xr-x   5 jp.pinob  staff  160 Dec 29 17:44 main
drwxr-xr-x  20 jp.pinob  staff  640 Dec 29 17:44 renderer

.webpack/main:
total 44896
drwxr-xr-x  5 jp.pinob  staff       160 Dec 29 17:44 adb
-rw-r--r--  1 jp.pinob  staff  11333523 Dec 29 17:44 index.js
-rw-r--r--  1 jp.pinob  staff  11650683 Dec 29 17:44 index.js.map

.webpack/renderer:
total 5792
-rw-r--r--  1 jp.pinob  staff   78268 Dec 29 17:44 1551f4f60c37af51121f.woff2
-rw-r--r--  1 jp.pinob  staff   89988 Dec 29 17:44 2285773e6b4b172f07d9.woff
-rw-r--r--  1 jp.pinob  staff  134294 Dec 29 17:44 23f19bb08961f37aaf69.eot
-rw-r--r--  1 jp.pinob  staff  747927 Dec 29 17:44 2f517e09eb2ca6650ff5.svg
-rw-r--r--  1 jp.pinob  staff  144714 Dec 29 17:44 4689f52cc96215721344.svg
-rw-r--r--  1 jp.pinob  staff   33736 Dec 29 17:44 491974d108fe4002b2aa.ttf
-rw-r--r--  1 jp.pinob  staff  133988 Dec 29 17:44 527940b104eb2ea366c8.ttf
-rw-r--r--  1 jp.pinob  staff   34034 Dec 29 17:44 77206a6bb316fa0aded5.eot
-rw-r--r--  1 jp.pinob  staff   13224 Dec 29 17:44 7a3337626410ca2f4071.woff2
-rw-r--r--  1 jp.pinob  staff  918991 Dec 29 17:44 7a8b4f130182d19a2d7c.svg
-rw-r--r--  1 jp.pinob  staff  203030 Dec 29 17:44 9bbb245e67a133f6e486.eot
-rw-r--r--  1 jp.pinob  staff   16276 Dec 29 17:44 bb58e57c48a3e911f15f.woff
-rw-r--r--  1 jp.pinob  staff  202744 Dec 29 17:44 be9ee23c0c6390141475.ttf
-rw-r--r--  1 jp.pinob  staff   76736 Dec 29 17:44 d878b0a6a1144760244f.woff2
-rw-r--r--  1 jp.pinob  staff  101648 Dec 29 17:44 eeccf4f66002c6f2ba24.woff
drwxr-xr-x  4 jp.pinob  staff     128 Dec 29 17:44 main_window
-rw-r--r--  1 jp.pinob  staff     854 Dec 29 17:44 main_window.334858a6200c99d614e3.hot-update.js
-rw-r--r--  1 jp.pinob  staff      35 Dec 29 17:44 main_window.334858a6200c99d614e3.hot-update.json
```

I think the problem is webpack can't find the dependency as in #463 but I tried the fix mentioned there and the error still occurs. When running the application I get the following error:
```
App threw an error during load
Error: No native build was found for platform=darwin arch=x64 runtime=electron abi=99 uv=1 libc=glibc node=16.9.1 electron=16.0.5 webpack=true
    loaded from: /Users/jp.pinob/Projects/pos-digital/carga-archivos-local

    at Function.load.path (/Users/jp.pinob/Projects/pos-digital/carga-archivos-local/.webpack/main/index.js:140689:9)
    at load (/Users/jp.pinob/Projects/pos-digital/carga-archivos-local/.webpack/main/index.js:140651:30)
    at Object../node_modules/usb/dist/usb/bindings.js (/Users/jp.pinob/Projects/pos-digital/carga-archivos-local/.webpack/main/index.js:249047:94)
    at __webpack_require__ (/Users/jp.pinob/Projects/pos-digital/carga-archivos-local/.webpack/main/index.js:274231:42)
    at Object../node_modules/usb/dist/usb/device.js (/Users/jp.pinob/Projects/pos-digital/carga-archivos-local/.webpack/main/index.js:249091:11)
    at __webpack_require__ (/Users/jp.pinob/Projects/pos-digital/carga-archivos-local/.webpack/main/index.js:274231:42)
    at Object../node_modules/usb/dist/usb/index.js (/Users/jp.pinob/Projects/pos-digital/carga-archivos-local/.webpack/main/index.js:249657:16)
    at __webpack_require__ (/Users/jp.pinob/Projects/pos-digital/carga-archivos-local/.webpack/main/index.js:274231:42)
    at Object../node_modules/usb/dist/webusb/index.js (/Users/jp.pinob/Projects/pos-digital/carga-archivos-local/.webpack/main/index.js:249967:11)
    at __webpack_require__ (/Users/jp.pinob/Projects/pos-digital/carga-archivos-local/.webpack/main/index.js:274231:42)
```

jp-pino, 2021-12-29:
Adding `node: {__dirname: true}` to my main webpack config seems to have fixed the issue.

mildsunrise, 2022-03-24:
This is a duplicate of #463, I think

thegecko, 2022-03-25:
Thanks, please re-open this if required

closed 2022-03-25 by thegecko