node-usb/node-usb #466: Using navigator.usb in the browser

gniezen, 2022-01-13:
Hi there!

With node-usb v2 now supporting the WebUSB API, would it maybe be possible to have the library use `navigator.usb` when available? I would love to be able to use the same code in Node.js, Electron and the browser, and at the moment I'm having to use what feels like a hacky workaround:

```js
let isBrowser, usb = null;

if (typeof navigator !== 'undefined') {
  const userAgent = navigator.userAgent.toLowerCase();
  isBrowser = userAgent.indexOf(' electron/') === -1 && typeof window !== 'undefined';
} else {
  // Node.js process
  isBrowser = false;
}

if (!isBrowser) {
  // For Node.js and Electron
  const { webusb } = await import('usb');
  usb = webusb;
} else {
  usb = navigator.usb; // Yay, we're using WebUSB!
}
```

Instead, I would just like to do an `import { webusb } from 'usb';` and have it use `navigator.usb` in the browser and `node-usb` in Node.js and Electron.

thegecko, 2022-01-13:
Hmm, interesting suggestion.

One drawback would be the user doesn't have control over which implementation is returned (but TBF, I can't think of an environment where both would be available).

thegecko, 2022-01-15:
Something like #467?

gniezen, 2022-01-19:
Yes, that looks like it will work just fine!

closed 2022-01-29 by thegecko