node-usb/node-usb #472: Does not support the ARM

dohooo, 2022-01-28:
First many thanks to the great contributors for making such an amazing project. Thank you for your efforts! 🍺

Reproduce steps:
1.  git clone https://github.com/dohooo/node-usb-reproduction.git
1.  yarn
1. yarn watch

You will then see this error on the console, The code is here `/Users/zhaodonghao/code/temp/vite-electron-builder/packages/preload/src/sha256sum.ts`

Logs:
[image]

[localhost-1643350529203.log](https://github.com/node-usb/node-usb/files/7956261/localhost-1643350529203.log)

Does it look like the ARM compiled version is not found??  This mistake gives me a headache and I hope I can get help

dohooo, 2022-01-28:
@thegecko 🙏

dohooo, 2022-01-28:
Related with https://github.com/electron/electron/issues/32662

thegecko, 2022-01-29:
Thanks for noticing this, it was an oversight

PR #473 (2022-01-29): Build universal binaries
Update gyp files to include flags for building universal darwin binaries

Fixes #472
closed 2022-01-29 by thegecko