node-usb/node-usb #475: Invalid memory read in configurationName

RReverser, 2022-02-03:
I have a Sony a6600, configured with Zadig on Windows to use WinUSB.

When queried via WebUSB from latest Chrome, the configurations look like this:

```js
> d.productName
'ILCE-6600'
> USBConfiguration {configurationValue: 1, configurationName: null, interfaces: Array(1)}
```

However, when queried via WebUSB implementation in node-usb v2.1.0, the same device's configurations return:

```js
> d.productName
'ILCE-6600'
> d.configurations[0]
{
  configurationValue: 1,
  configurationName: 'Љ',
  interfaces: [
    {
      interfaceNumber: 0,
      alternate: [Object],
      alternates: [Array],
      claimed: false
    }
  ]
}
```

I suspect something in node-usb is not handling missing `configurationName` correctly, and ends up reading invalid memory.

RReverser, 2022-02-03:
Same happens in other null-ish names, e.g. in alternates.

Chrome:

```js
> d.configurations[0].interfaces[0].alternates[0]
USBAlternateInterface {alternateSetting: 0, interfaceClass: 6, interfaceSubclass: 1, interfaceProtocol: 1, interfaceName: null, …}
```

Node:

```js
> d.configurations[0].interfaces[0].alternates[0]
{
  alternateSetting: 0,
  interfaceClass: 6,
  interfaceSubclass: 1,
  interfaceProtocol: 1,
  interfaceName: 'Љ',
  ...
}
```

thegecko, 2022-03-20:
TBF, I see the same behaviour with chrome on Mac OS ('Љ' appearing for empty strings). I fear this is an issue far lower in the stack.

A quick google shows this turn up in other usb implementations, too :/

kevinmehall, 2022-03-20:
String descriptor index 0 is supposed to be a list of supported languages rather than a string. 0x0409 is the language code for US English. 0x0409 as a character in UTF16-LE is `Љ`. So it should interpret string descriptor index 0 as null instead making a GET DESCRIPTOR request.

thegecko, 2022-03-20:
> So it should interpret string descriptor index 0 as null instead making a GET DESCRIPTOR request.

Ooh I wasn't aware of that, I'll create a PR...

thegecko, 2022-03-20:
Fix in #487, does that look right?

kevinmehall, 2022-03-21:
Yep :+1:

PR #487 (2022-03-21): getStringDescriptor 0 returns null
Shortcut `getStringDescriptor()` to return undefined when passed index 0.

Fixes #475
closed 2022-03-21 by thegecko