node-usb/node-usb #476: LIBUSB_ERROR_IO on .controlTransfer

ColPaulR, 2022-02-07:
I am trying to send 3 bytes to a device to set the LCD display to the values I want. When I sniff the usb protocol with BUSDOG, I see the 3 bytes go across from another program. When I try to replicate with the below code, it runs but does not actually send the control transfer. When I step into .controlTransfer, everything works until the error exception is caught at line 175. The error message is LIBUSB_ERROR_IO. This is on a relatively fresh Windows 10 installation, node v16.13.2,  npm 8.1.2. usb@2.1.0. Attached is the code snippet renamed from .js to .txt so i could attach. Please help!
 
[test_controltransfer.txt](https://github.com/node-usb/node-usb/files/8011753/test_controltransfer.txt)

kuangping-kp, 2022-07-21:
问题解决了？

thegecko, 2022-07-21:
It doesn't look like you are waiting for the control transfers to finish before running another one. Use the callback to chain your calls or promisify the transfers and await them.

If you are accessing a HID device, recommend you try https://github.com/node-hid/node-hid

closed 2023-02-26 by thegecko