node-usb/node-usb #480: missing libudev on raspbian?

pelikhan, 2022-02-10:
It seems that libudev-dev is not installed by default on raspbian; as a result, node-usb fails to install.

```
make: Entering directory '/home/pi/gh/jacdac-cli/node_modules/usb/build'
  CC(target) Release/obj.target/libusb/libusb/libusb/core.o
  CC(target) Release/obj.target/libusb/libusb/libusb/descriptor.o
  CC(target) Release/obj.target/libusb/libusb/libusb/hotplug.o
  CC(target) Release/obj.target/libusb/libusb/libusb/io.o
  CC(target) Release/obj.target/libusb/libusb/libusb/strerror.o
  CC(target) Release/obj.target/libusb/libusb/libusb/sync.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/poll_posix.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/threads_posix.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_usbfs.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_udev.o
../libusb/libusb/os/linux_udev.c:40:10: fatal error: libudev.h: No such file or directory
   40 | #include <libudev.h>
      |          ^~~~~~~~~~~
compilation terminated.
make: *** [libusb.target.mk:154: Release/obj.target/libusb/libusb/libusb/os/linux_udev.o] Error 1
make: Leaving directory '/home/pi/gh/jacdac-cli/node_modules/usb/build'
```

I installed it manually and run the install again and things worked. 
```
sudo apt install libudev-dev
```

I wonder if this kind of dependencies can be baked into the package.json.

thegecko, 2022-02-10:
`libudev-dev` is a compilation requirement which would be difficult to include.

I'm more concerned why the prebuilt binaries don't work (RPi should use the arm-linux prebuilds)

thegecko, 2022-02-10:
I've just tested `v2.1.2` on rpi and rpi64 with no issue. They use the shipped prebuild binaries.

There should be no need to compile the package, are you trying to rebuild the package with `electron-rebuild` or similar?

pelikhan, 2022-02-10:
I was doing a yarn install on my package that uses usb v2.1.2.

I did run apt update/upgrade before that. Maybe I need a new image.

thegecko, 2022-02-10:
TBF, this is all testing via a remote CLI, but I doubt running it through raspbian should make a difference.

mildsunrise, 2022-03-24:
This issue should be closed, as libudev should never be bundled statically AFAIK.
Instead we should add instructions to the README telling users to install `build-essential libudev-dev` if the prebuilds don't work.

And yes it is normal for prebuilds not to work, especially on Raspis and other embedded devices, because of libc compatibility issues:

~~~
Error: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by /home/user/app/node_modules/usb/prebuilds/linux-arm64/node.napi.armv8.node)
~~~

In that case it will fall back to compilation and you'll need libudev-dev.
If I find time, I'll try to make prebuilds target an older glibc version to make them more portable.

thegecko, 2022-03-25:
I'll close this issue as I agree libudev-dev is an end user concern.

@mildsunrise , if you want to look at the prebuild images, they are maintained here:

https://github.com/node-usb/docker-images

closed 2022-03-25 by thegecko

pelikhan, 2022-03-25:
A little note in the readme would be awesome! Thanks for digging into this.

https://github.com/node-usb/docker-images<https://nam06.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fnode-usb%2Fdocker-images&data=04%7C01%7Cjhalleux%40microsoft.com%7Cd84cc8e0fd494fa784a508da0e703e45%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C637838172491831270%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000&sdata=%2BGP0tYhakaAAEYc3KqbVOV6FYN6uo8h6E6UEd%2Fw2BRM%3D&reserved=0>