node-usb/node-usb #481: Attaching usb doesn't trigger usb.on('attach') event

YiyuanYin, 2022-02-14:
I have an electron application that use this package to listen to the usb attach event. Sometimes attaching a usb can trigger `usb.on('attach')` event, sometimes it won't.
The code checking whether a usb is attached is by finding the new device which has the different `deviceAddress` from the former one. However, sometimes when a usb is attached, the `deviceAddress` is the same as the old ones in the device list. Therefore, `usb` won't emit `attach` event.
[image]
Have you ever thought about adding some other conditions to decide whether a usb is attached?

thegecko, 2022-02-19:
> The code checking whether a usb is attached is by finding the new device which has the different deviceAddress from the former one.

The attach/detach events are exposing hotplug events from the [libusb](https://github.com/libusb/libusb) library (except on Windows)

Perhaps you need to raise this issue there?

mildsunrise, 2022-03-25:
You mentioned "The code checking whether a usb is attached" so I assume you're in Windows.
There's #488 to try to implement actual hotplug detection support on Windows.

mildsunrise, 2022-03-27:
By the way, I've just realized why this doesn't work... `deviceAddress` is only unique *within the bus the device is connected to*.
To properly identify the device, we need the `busNumber` too. Something like the `deviceId` from WebUSB:

https://github.com/node-usb/node-usb/blob/c42aa55fba09ea65c660d78cdd3779cb29f1d04c/tsc/webusb/index.ts#L347-L353

But I think we can do it even simpler... libusb will never return different `libusb_device` pointers for the same device. And the bindings will never create two `Device` instances for the same `libusb_device`. Therefore we should be able to just say `device1 === device2` to compare both lists of devices.

closed 2023-02-26 by thegecko