node-usb/node-usb #483: LIBUSB_ERROR_ACCESS while calling device.open()

Kainos-SU, 2022-02-25:
Hello, it is my first time I post an issue on github, plus I have a poor knowledge of English, so I am sorry.
I have computer on Linux Mint and thermal USB printer CUSTOM, vendorId = 0xdd4. When i try to found and open device programm fails whith error `LIBUSB_ERROR_ACCESS`. But I found thet when I run before `sudo chmod -R 777 /dev/bus/usb/` then error not apear.
I work first time whith USB devices. Can you help me is there a way to access to USB device without such "solutions"?

thegecko, 2022-02-25:
Please look into udev rules when using usb devices in Linux

Kainos-SU, 2022-02-25:
I am sory, but I not wery familiar with USB at all, can you guide me to sources where I can learn how to configure `udev` for my purposes?
And also can I make this by my software?

closed 2023-02-26 by thegecko