node-usb/node-usb #488: Use native usb device detection on Windows

thegecko, 2022-03-21:
The current device detection on windows uses a polling mechanism from JavaScript which feels wasteful and dirty.
I plan to try merging the native code from the [node-usb-detection library](https://github.com/MadLittleMods/node-usb-detection) to undertake this in a cleaner way.

Rough plan could be:

- [x] Add native source into node-usb and get it compiling in the CI
- [ ] Wire in the detection start/stop functions into `_enableHotplugEvents` and `_disableHotplugEvents`
- [ ] Find a way of uniquely identifying a device in the detection code in order to:
- [ ] Load the device using libusb and raise the relevant event from node-usb

To this end, I now have the NAPI port of `node-usb-detection` from @Julusian sitting on the [windows-native-detection branch](https://github.com/node-usb/node-usb/tree/windows-native-detection)

Julusian, 2022-07-18:
I've made a start on this, but am having a hard time translating the windows events into libusb devices.
I am half way through exploring a way to translate, but I am wondering if it is enough for it to report that 'something changed' and use that to trigger a poll.
I don't see a way of getting a `libusb_device` without using `libusb_get_device_list`, so the difference is really whether a diff needs to be done, or if the backend can provide ids to say which device changed (with the hope it wont have disappeared before `libusb_get_device_list` is called). I am starting to feel like it isnt worth the extra effort to do the ids method, and a diff when we know something has changed should be good enough.

thegecko, 2022-07-18:
>  I am wondering if it is enough for it to report that 'something changed' and use that to trigger a poll.

That would be a lot better than the current polling mechanism and I suspect a changed event may well be enough. Do you want to open a PR for discussion and we can grow from there perhaps?

PR #524 (2022-07-30): feat: native usb device detection on Windows
## Changes

This is the start of integrating native usb detection.
As there are now two hotplug backends, I split the different implementations into their own files.

I havent found a good way to translate from the path that the detection provides to identifiers that could be matched on a `libusb_device`. And as the only way to obtain a `libusb_device` is through `libusb_get_device_list`, I am unsure if it is worth the extra work.

For this I have added new `attachIds` and `detachIds` events, that report the vendorId and productId of the device that [... cut]
closed 2022-07-30 by thegecko