node-usb/node-usb #491: Make the addon context-aware

mildsunrise, 2022-03-25:
It would be great if we could make node-usb [context aware](https://nodejs.org/api/addons.html#context-aware-addons), i.e. so it can be initialized multiple times from different Node.js instances (node_envs). (Actually the addon incorrectly declares itself as context aware, and that's why users get crashes)

This way it can safely run in Electron and worker threads, probably fixing: #458 #378

This basically requires global variables to be [moved to environment-specific data](https://nodejs.org/api/n-api.html#environment-life-cycle-apis).

> <details><summary>Required changes</summary>
> I've reviewed the code to see the variables we should move, let me know if I'm missing something.
> 
> At least the following static variables in `Device` (I think node-addon-api already saves constructor for us):
> 
> https://github.com/node-usb/node-usb/blob/007e702c3053f4cf649356b4cb3994595a82885e/src/node_usb.h#L70-L71
> 
> And the following hotplug global variables:
> 
> https://github.com/node-usb/node-usb/blob/007e702c3053f4cf649356b4cb3994595a82885e/src/node_usb.cc#L159
> 
> https://github.com/node-usb/node-usb/blob/007e702c3053f4cf649356b4cb3994595a82885e/src/node_usb.cc#L191-L193
> 
> And because each context will use different threads, this as well (or protect with a mutex):
> 
> https://github.com/node-usb/node-usb/blob/007e702c3053f4cf649356b4cb3994595a82885e/src/node_usb.cc#L26
> 
> I think it would be a good idea to use a different libusb context for each Node.js context, so this too:
> 
> https://github.com/node-usb/node-usb/blob/007e702c3053f4cf649356b4cb3994595a82885e/src/node_usb.cc#L15
> </details>

As an implication, we also need to implement cleanup (make the addon unloadable) with a [cleanup hook](https://nodejs.org/api/n-api.html#napi_add_env_cleanup_hook). In Windows, we need to tell the thread to exit + join it. Then the tests.

I'm **not** saying I'll have time to do this, just opening this to track the issue (and point people complaining about Electron / workers here) and to see what your thoughts are.

mildsunrise, 2022-03-25:
@thegecko thanks for taking on this. if I ever have time to work on it myself, I'll hit you up.
Looking more at the code, I think the most complicated part is going to be the cleanup.

mildsunrise, 2022-03-25:
by the way, we may want to wait until v1.0.26 is released, which fixes a bug regarding cleanup, see https://github.com/libusb/libusb/commit/c3639bc23eaaed7fa94d377af42f00ddc5083de2

thegecko, 2022-04-09:
Looking into this and unravelling the layers, I think the simplest, cleanest and most reliable approach is to use the [node-addon-api addon structure](https://github.com/nodejs/node-addon-api/blob/main/doc/addon.md).

This uses `napi_set_instance_data`/`napi_get_instance_data` under the hood which would need an increase in the NAPI version from 4 to 6 (Node.js >= v10.20.0 or >= v12.17.0).

I assume any other approach to keep the NAPI version at 4  would be a lot more involved?

mildsunrise, 2022-04-09:
> I think the simplest, cleanest and most reliable approach is to use the [node-addon-api addon structure](https://github.com/nodejs/node-addon-api/blob/main/doc/addon.md).

The cleanest way as you say is `Napi::Addon`.
If you don't want to refactor all the code into a single class, `env.GetInstanceData()` with a struct works just as well (it's just a bit more verbose to access instance data). [Example (the constructor isn't necessary).](https://github.com/mildsunrise/node_netlink/commit/e2410a9be9bceaaf33231dc0386fe70985dcb7de)
It's also easier to polyfill if we want to preserve support with v10, see below.

> This uses napi_set_instance_data/napi_get_instance_data under the hood which would need an increase in the NAPI version from 4 to 6 (Node.js >= v10.20.0 or >= v12.17.0).

ah, true. The proper way to make this addon context-aware is using `napi_set_instance_data` / `napi_get_instance_data`, which would require dropping support for v10 and requiring v12, but given that [v10 reached EOL long ago, and v12 will in a few days](https://nodejs.org/en/about/releases), I think that's reasonable?

If you would like to preserve support, `napi_get_instance_data` can be emulated by saving the instance data in the JS module object, with an ObjectWrap like always, and making sure it gets passed to the global callbacks + constructors. It's more cumbersome but shouldn't make a significant difference.

thegecko, 2022-04-24:
I've had a crack at fixing this over at #512

@mildsunrise, it builds and passes all the unit tests, but I'd really appreciate a review if you have some time?

Also, I can't see where a cleanup hook would be useful, from what I've read it seems the state is entirely handled by `node-addon-api`.

Finally, any ideas on how to create tests for this and prove it's context safe would be welcomed.

mildsunrise, 2022-04-25:
That's some nice work, thanks! I'll review when I find some time.

Yes, tests are important for this. It shouldn't be hard to create scripts that crash the current (non-context aware) version, simply spin up a [worker thread](https://nodejs.org/docs/latest/api/worker_threads.html) and use the library from both the main instance and the worker. Subscribing to hotplug, calling getDeviceList, and opening a device (preferably the same device) from both sides, I think that should suffice.

mildsunrise, 2022-04-25:
Regarding:

> Also, I can't see where a cleanup hook would be useful, from what I've read it seems the state is entirely handled by node-addon-api.

I've left a comment in the review detailing the two basic cleanup actions we should take when the module is unloaded.
These cleanup actions can be done inside of a [registered cleanup hook](https://github.com/nodejs/node-addon-api/blob/main/doc/env.md#addcleanuphook), or *maybe* it's also allowed to do them when the instance data is destructed... I have to check, though.

mildsunrise, 2022-04-25:
I'd like to do more stuff (such as the tests) but I'll be beginning a new job (at Dolby) very soon, so I don't think I'll have much time :(

thegecko, 2022-04-26:
> I'll be beginning a new job (at Dolby) very soon

Nice!

thegecko, 2022-05-07:
Released in v2.4.0

closed 2022-05-07 by thegecko