node-usb/node-usb #494: v2.2.0 segfaults on Arch Linux

ottomated, 2022-03-25:
After upgrading to v2.2.0, the following code sometimes segfaults on Arch Linux, killing the entire node process:
```js
const { WebUSB } = require('usb');
const usb = new WebUSB({ allowAllDevices:true })
usb.getDevices().then(console.log).catch(console.error)
```
After reverting to 2.1.3, the same code functions properly.

mildsunrise, 2022-03-25:
hmm, in that case it has to be because of https://github.com/node-usb/node-usb/pull/492... let me check

mildsunrise, 2022-03-25:
ah no, it bisects to the upgrade to libusb v1.0.25 (#490).
it seems to be libusb/libusb#1059, which is popping up in other apps and distros as well.
the bug was fixed in https://github.com/libusb/libusb/pull/1073 (https://github.com/libusb/libusb/commit/7cc06ea5c7c3b36801421a6be17b51b92c1bc05a) but is still unreleased.

I can confirm that applying this fix solves the issue. like I said it hasn't landed in a release yet, but they're preparing to release v1.0.26 in a few days (libusb/libusb#1047). since it's a fix only release, I think it's better to wait for it.

thegecko, 2022-04-02:
Looks like the v1.0.26 release has been pushed to April 10, 2022

sergeu90, 2022-04-03:
@thegecko Maybe downgrade libusb version?

mildsunrise, 2022-04-03:
I'd vote for incorporating the current v1.0.26 rc than downgrading. In the future if there are further changes since the rc, we can bring them as well.

thegecko, 2022-04-11:
v2.3.0 has been released, please check this fixes your issue.

closed 2023-02-26 by thegecko