node-usb/node-usb #495: LIBUSB_ERROR_NO_DEVICE

seb-mtl, 2022-03-27:
We use `libusb` to detect attach/detach events of USB devices in our Electron app. We recently came across two users that reported the error below.

I'm not sure if this is caused by registering the event handlers, or if this caused by loading the module in the first place. Can anyone shed some lights on this?

[image]

mildsunrise, 2022-03-27:
I think it's a race. a while ago, we replaced the native hotplug patch with an emulation by periodically polling the output of `getDeviceList()` and taking differences:

https://github.com/node-usb/node-usb/blob/54c6e5dfc4cc990d278805b7ed6d32825b17479d/tsc/usb/index.ts#L60-L74

this `deviceAddress` isn't a property, it's a getter that calls `libusb_get_device_address`. it's possible for a device to appear in `getDeviceList` and go away just before calling `libusb_get_device_address`, therefore resulting in a `NO_DEVICE` error. we should handle that error correctly.

YiyuanYin, 2022-04-25:
I came cross with the same problem,
[image]

thegecko, 2022-04-25:
Is this still occurring since v2.3.0?

closed 2023-02-26 by thegecko