node-usb/node-usb #497: allow unreffing hotplug

mildsunrise, 2022-03-27:
little feature request, very low priority, just here to remind myself in case I have time someday.

it would be cool if `usb` had `ref()` and `unref()` methods (like all Node.js event sources) to allow not holding the event loop alive, for projects that need it.

thegecko, 2022-04-02:
Would that be similar to the hotplug ref/unref but for the [usb_context](https://github.com/node-usb/node-usb/blob/master/src/node_usb.cc#L15)?

mildsunrise, 2023-02-27:
sorry, I can't make sense of this issue... as you mentioned, there's ref/unref for the hotplug events.
according to the history, it already existed when I opened this issue. really weird. anyway, closing.

closed 2023-02-27 by mildsunrise