node-usb/node-usb #498: WebUSB layer mutex issues

phidgets, 2022-03-30:
The WebUSB layer adds a mutex which locks around every call into the device.

First, the mutex is written such that a lock awaiter won't always acquire the lock next.

If I poll an endpoint in a loop:

for (;;) {
  const data = await dev.transferIn(1, 64);
  processData(data);
}

And then somewhere else I want to write to an endpoint:

await dev.transferOut(2, buf);

The second transfer just waits forever, because the loop is internally unlock() then lock() the mutex, but there isn't any sort of yield to lock awaiters.

The second issue, is that this kind of locking shouldn't be happening at all. I should be able to await a transferIn, and while that is waiting, also do a transferOut. transferIn() can block indefinitely if the USB device is just NAKing on that endpoint. As written, if I start a transferIn() for an endpoint that's NAKing, that locks up the whole device, so that it can't be communicated with at all, or even closed.

WebUSB in the browser works properly

mildsunrise, 2022-03-31:
This was something I also wanted to work on. Even if the mutex was fair and did FIFO or something, it would still defeat the point because you'd only be able to have 1 simulataneous transfer on a device.

On I/O heavy applications, it's standard practice to keep multiple active transfers with the device at all times to get good bandwidth.

I remember I tried to `git blame` to figure out if the mutex was added because of a particular reason, but it seems to be there from day one. @thegecko was it put in place out of general safety concerns or did you experience issues that made you place it?

thegecko, 2022-04-02:
> @thegecko was it put in place out of general safety concerns or did you experience issues that made you place it?

I wrote the WebUSB device wrapper for a different project and can't remember why the mutex was added. Memory isn't what it used to be!

I've removed this over in https://github.com/node-usb/node-usb/pull/501 and all the tests still seem to pass.

Shall I create a test release for you to try?

thegecko, 2022-04-11:
Mutex is removed in v2.3.1, please confirm this resolves your issue.

closed 2022-04-11 by thegecko