node-usb/node-usb #499: Intermittent Failure during `getDeviceList` call

noah-moutfitters, 2022-04-01:
I am currently experiencing an issue where there is an intermittent error being thrown when calling `getDeviceList`. The error thrown is `Error in native callback`. I am unable to identify where and why the error is thrown. Any suggestions would be helpful.

roman-veryovkin, 2022-04-02:
I have the same issue. Any help would be pretty much appreciated as this problem is very annoying.

sergeu90, 2022-04-02:
I have some issue

thegecko, 2022-04-02:
Could you confirm the version this issue is appearing in and whether it's still an issue in v2.1.3

sergeu90, 2022-04-02:
Yes. It is on last version

thegecko, 2022-04-02:
Theres been some larger changes in the latest release, can you confirm if you still see the issue if you roll back to `v2.1.3`?

sergeu90, 2022-04-02:
Yes. My app crash sometimes on 2.1.3 also. Our code connect to the thermal printer and send command. After update on last verstion we think problem gone but sometimes we have crash app. 

https://github.com/node-usb/node-usb/issues/500

in this ticket i attached crash report.

thegecko, 2022-04-02:
Thanks @sergeu90 I think you may be seeing a different issue to the OP.

@noah-moutfitters would you mind trying `v.2.1.3` too?

noah-moutfitters, 2022-04-02:
@thegecko thanks for looking into this. My current version is `v.2.1.2` according to my lock file.

noah-moutfitters, 2022-04-02:
And I will look into that tomorrow, currently not at computer.

sergeu90, 2022-04-02:
@thegecko  yes it is on v.2.1.2  also

noah-moutfitters, 2022-04-03:
@thegecko I know you didn't suggest this but upgrading to `v2.2` solved all of my issues. Weirdly, even a serial port one(although this one occurred even more sparsely, so I am still not sure). I am good with closing this if you are content with solution.

closed 2022-04-03 by noah-moutfitters