node-usb/node-usb #500: Electron App crash.

sergeu90, 2022-04-02:
Hello everyone. I have some strange problem. I use webusb and sometimes my app crash. 

`

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               Electron Helper (Renderer) [99466]
Path:                  /Users/USER/*/Electron.app/Contents/Frameworks/Electron Helper (Renderer).app/Contents/MacOS/Electron Helper (Renderer)
Identifier:            com.github.Electron.helper
Version:               ???
Code Type:             X86-64 (Native)
Parent Process:        Electron [93391]
Responsible:           webstorm [78579]
User ID:               501

Date/Time:             2022-04-02 15:57:58.4093 +0300
OS Version:            macOS 12.2.1 (21D62)
Report Version:        12
Bridge OS Version:     3.0 (14Y910)
Anonymous UUID:        49D68D9F-851C-B63E-5D5B-B3DE0E9F648D

Sleep/Wake UUID:       DDDA2746-9B29-4F7B-BD1C-66B3558D3A9E

Time Awake Since Boot: 670000 seconds
Time Since Wake:       21898 seconds

System Integrity Protection: enabled

Crashed Thread:        0  CrRendererMain  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000091f2a
Exception Codes:       0x0000000000000001, 0x0000000000091f2a
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [99466]

VM Region Info: 0x91f2a is not in any region.  Bytes before following region: 4484931798
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      10b5bc000-10b5de000    [  136K] r-x/r-x SM=COW  ...er (Renderer)

Thread 0 Crashed:: CrRendererMain Dispatch queue: com.apple.main-thread
0   node.napi.node                	       0x10ea0c9ff libusb_free_transfer + 21
1   node.napi.node                	       0x10ea05594 Transfer::~Transfer() + 66
2   node.napi.node                	       0x10ea05666 Transfer::~Transfer() + 14
3   node.napi.node                	       0x10ea06a59 Napi::ObjectWrap<Transfer>::FinalizeCallback(napi_env__*, void*, void*) + 55
4   Electron Framework            	       0x11e5eb6e7 node_api_get_module_file_name + 5207
5   Electron Framework            	       0x11e5cd5a7 node::EmitAsyncDestroy(node::Environment*, node::async_context) + 214823
6   Electron Framework            	       0x11e5cce18 node::EmitAsyncDestroy(node::Environment*, node::async_context) + 212888
7   Electron Framework            	       0x11e5cc890 node::EmitAsyncDestroy(node::Environment*, node::async_context) + 211472
8   Electron Framework            	       0x117f70c04 uv_check_start + 260
9   Electron Framework            	       0x117f6b000 uv_run + 336
10  Electron Framework            	       0x1180ce963 v8::CodeEvent::GetScriptName() + 198019
11  Electron Framework            	       0x11ada524e node::AsyncResource::get_async_id() const + 8500126
12  Electron Framework            	       0x11adc3049 node::AsyncResource::get_async_id() const + 8622489
13  Electron Framework            	       0x11ae02695 node::AsyncResource::get_async_id() const + 8882149
14  Electron Framework            	       0x11ae01802 node::AsyncResource::get_async_id() const + 8878418
15  Electron Framework            	       0x11ae01dcf node::AsyncResource::get_async_id() const + 8879903
16  CoreFoundation                	    0x7ff8180738fd __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
17  CoreFoundation                	    0x7ff818073865 __CFRunLoopDoSource0 + 180
18  CoreFoundation                	    0x7ff8180735e4 __CFRunLoopDoSources0 + 242
19  CoreFoundation                	    0x7ff81807201b __CFRunLoopRun + 893
20  CoreFoundation                	    0x7ff8180715dd CFRunLoopRunSpecific + 563
21  Foundation                    	    0x7ff818ed866e -[NSRunLoop(NSRunLoop) runMode:beforeDate:] + 216
22  Electron Framework            	       0x11ae02e39 node::AsyncResource::get_async_id() const + 8884105
[... 515 lines cut ...]
      },
      "deploymentId" : 240000128
    },
    {
      "rolloutId" : "5fc94383418129005b4e9ae0",
      "factorPackIds" : {

      },
      "deploymentId" : 240000313
    },
    {
      "rolloutId" : "5fb4245a1bbfe8005e33a1e1",
      "factorPackIds" : {

      },
      "deploymentId" : 240000014
    }
  ],
  "experiments" : [

  ]
}
}

`

thegecko, 2022-04-02:
Looks like a crash in libusb after a transfer. @mildsunrise any idea?

sergeu90, 2022-04-02:
it is issue reproduce randomize. i can click button many times and all good but sometimes app crash on our clients and developement

sergeu90, 2022-04-02:
On mac os issue reproduce each time. when i press btn find devices and after i selected device and attempt to do repeat action and app crash. but on windows randomize

sergeu90, 2022-04-02:
I noticed when a refresh app a get same crash app with libusb in log

roman-veryovkin, 2022-04-03:
+1 I am experiecing the same issue in my app. Any ideas how to workaround it?

sergeu90, 2022-04-03:
@thegecko Hi. I rechecked on 2.1.3 and my mac os app not crashed but i have this error. 
[image]

mildsunrise, 2022-04-04:
This is the exact issue reported in #494. You are using v2.2, right? It's a libusb bug, see that issue for context and pending actions, in the meantime you can downgrade to v2.1.x.

mildsunrise, 2022-04-04:
@thegecko maybe we should create an issue template asking people to report Node.js/Electron version, usb version, and OS? People don't tend to think about it and it would be useful info.

thegecko, 2022-04-04:
Agreed, PRs always welcome ;)

sergeu90, 2022-04-04:
> This crash is the exact issue reported in #494. You are using v2.2, right? It's a libusb bug, see that issue for context and pending actions, in the meantime you can downgrade to v2.1.x.

If we downgrade to 2.1.* then we have error invalid argument

thegecko, 2022-04-04:
> @thegecko maybe we should create an issue template

Opened in #504

mildsunrise, 2022-04-04:
> If we downgrade to 2.1.* then we have error invalid argument

This is also expected since the fix (#492) only landed in v2.2.x.
You'll have to wait a bit until v2.2.1 if you need both fixes.

sergeu90, 2022-04-05:
> > If we downgrade to 2.1.* then we have error invalid argument
> 
> 
> 
> This is also expected since the fix (#492) only landed in v2.2.x.
> 
> You'll have to wait a bit until v2.2.1 if you need both fixes.

After release libusb. Do you publish update library?

mildsunrise, 2022-04-05:
That's the plan, yes. libusb v1.0.26 is scheduled to be released on April 10th, but we may release the current rc earlier.

sergeu90, 2022-04-06:
@mildsunrise @thegecko  When you planning update lib usb to rc version or you wait release ?

sergeu90, 2022-04-10:
https://github.com/libusb/libusb/releases/tag/v1.0.26 
@mildsunrise @thegecko I see lib usb released

mildsunrise, 2022-04-10:
I didn't create any PR because I saw @thegecko create the `libusb-1.0.26` branch with the rc, earlier.
There have only been 2 noteworthy changes since the RC: fix for isochronous transfers on Windows (https://github.com/libusb/libusb/commit/bfa8b8535a929e939d549d9c3778592abfafe870, https://github.com/libusb/libusb/commit/bfbef179b12110d8d8fc6ff9e4af5f96a37a78ab), fix for crash on MacOS when using multiple contexts (https://github.com/libusb/libusb/commit/ff05357a839dd1f65caf736ded2476eaa135fe06).

Should be safe to merge here and release a patch version. WDYT?

thegecko, 2022-04-11:
v2.3.0 has been released, please check this fixes your issue.

sergeu90, 2022-04-11:
@thegecko Works as excepted. Good work thanks

closed 2022-04-11 by sergeu90