node-usb/node-usb #502: printer cannot be used by other app

dadinugroho, 2022-04-02:
Hi All, 

I have a case where I wanted to use https://github.com/ThinhVu/tsc-printer-js. This require to use this package. then, I need to run zadig to install winusb driver. This works perfect, until I wanted to use my printer with other label software. My printer is not detected. 

How can I still use the printer after I used zadig to installed winusb driver.

noah-moutfitters, 2022-04-03:
@dadinugroho Speaking from personal experience, you may fail at this. Do you know what your original driver was?

mildsunrise, 2022-04-04:
I don't use Windows so correct me if wrong, but... it only allows one driver to handle a USB device, right? If instead of your normal driver you set it to use Zadig, then 'normal' applications can't use it since it doesn't appear as a printer. You'll have to install the original driver back for that. 

To my knowledge, Windows is the only OS with this limitation. Linux, Macos, etc should be fine.

@thegecko we keep getting these questions, maybe we should add a disclaimer to the README?

thegecko, 2022-04-04:
Yeah, this is a libusb on Windows usage issue really. If your device doesn't have a winusb driver you need to use zadig or the usbdk backend and both approaches claim exclusive access AFAICT

closed 2022-04-04 by thegecko

mildsunrise, 2022-04-04:
> If your device doesn't have a winusb driver

Ah okay, that was the part I was missing. So it depends on support from the original driver, I see.

dadinugroho, 2022-04-05:
Thank you @mildsunrise @thegecko . I would just want to confirm that. In my situation, it is a bit difficult and impractical should I need to keep changing the driver. 

The other issue with using zadig is that it cannot rolled back the old driver. I need to install the driver again from the beginning. Or, it is just me that does not know how to revert back the driver easily?

It is true I am using windows in my client side. Maybe I should setup a print server using linux and attached the printer and not keep changing the driver.

mildsunrise, 2022-04-05:
I don't know much about Windows but again, if it's a Linux box then it should work :)

PR #503 (2022-04-04): Add more Windows driver details to README
Closes #502