node-usb/node-usb #516: USB detection loop

adriencarbonaro, 2022-05-23:
## Bug Description:

When subscribing to `attach` event, the callback triggers in a loop.

## Steps to Reproduce:
1. Launch the following script with `ts-node` or after compiling it :
```
import * as usb from "usb";

usb.usb.on('attach', (device) => {
	console.log(device);
});
```
3. Check the logs. It displays a lot of attached devices, never stopping.

## Additional Information
- Operating System: Windows 10
- Runtime: `node@14.16.1`
- Node USB Version: `usb@2.4.1`

## EDIT

It works with `usb@2.3.1`, and doesn't starting from `usb@2.4.0`

thegecko, 2022-05-23:
Looks like this was introduced with https://github.com/node-usb/node-usb/pull/496

The devices returned from `getDeviceList()` aren't the same objects each time so `Set.has()` will always be false.

cc @mildsunrise

mildsunrise, 2022-05-24:
libusb seems to be returning multiple `libusb_device*` referring to the same USB device.
I've reported this upstream.
Maybe I was mistaken, and the "1 USB device = 1 `libusb_device*`" thing isn't a hard guarantee, let's wait for libusb to confirm if this is indeed their bug, or if we should perform additional checks.

mildsunrise, 2022-05-24:
ah, true. @adriencarbonaro said it works correctly (modulo the 2 attach events) on 2.3.1, so in principle it can't be #496.
could it be because of the context-aware changes?
I'll try to find a windows machine to hopefully reproduce this on, but I'm not sure when that will be

thegecko, 2022-05-24:
I can confirm it works as expected in `v2.3.1`, so I think it is due to the context-aware rewrite.

You can recreate the issue on any OS by setting `hotplugSupported` to false:

https://github.com/node-usb/node-usb/blob/master/tsc/usb/index.ts#L46

thegecko, 2022-05-24:
Yup, reverted https://github.com/node-usb/node-usb/commit/12db6e4acd9c4a3a5c388dd0d2aa0b8139dfd6b9 and it works again.

I'll close the other PR but have no idea what's causing the issue, but it's not `libusb`.

mildsunrise, 2022-05-25:
Okay. I won't close the upstream issue though, since I'd like to be sure the hotplug polyfill is being done correctly.

closed 2022-05-25 by thegecko

thegecko, 2022-05-27:
Fixed in `v2.4.2`

PR #519 (2022-05-25): fixes in context-aware (#512)
Aaagh, such a silly error. We were copying the byPtr map instead of obtaining a reference to it.

Refs #512
Fixes #516
commit d69af409 (2022-05-25): fixes in context-aware (#512) (#519)

Aaagh, such a silly error. We were copying the byPtr map instead of obtaining a reference to it.

Fixes #516