node-usb/node-usb #518: electron build windows exe, open exe error

CahierX, 2022-05-24:
## Bug Description:
electron build windows exe , open exe error
native Node module causes error: Uncaught Error: No native build was found for platform=win32 arch=x64 runtime=electron abi=99 uv=1 libc=glibc node=16.9.1 electron=16.0.4 webpack=true 

## Additional Information
Windows 10 
Node 16.9.1
electron 16.0.4
[image]

thegecko, 2022-05-27:
Please confirm this example works on your system:

https://github.com/node-usb/node-usb-example-electron

apurvamaloo, 2022-06-12:
I am also getting similar error while packaging the electron app : 

`Error: No native build was found for platform=win32 arch=x64 runtime=electron abi=85 uv=1 libc=glibc node=16.13.2 electron=18.3.3 webpack=true`

[image]

I created a sample app which is using usb at https://github.com/apurvamaloo/node-usb-sample-app

Note: `npm start` works fine but `npm run package`  is throwing error

thegecko, 2022-06-13:
> I created a sample app which is using usb

That's quite a large sample app. I was unable to compile it with Node 12, 14 or 16

thegecko, 2022-06-13:
I've tested the latest release with the latest Electron version (v19) on Windows and mac and cannot reproduce this issue.

Please recreate this problem based on the example app:
https://github.com/node-usb/node-usb-example-electron

Things which can cause this problem:

- Forcing a rebuild of the usb package (e.g. `electron-rebuild usb`), it isn't required as it uses N-API
- Loading the USB library into the frontend process
- relying on something else to manage your packages

The webpack=true aspect here is interesting. Perhaps try packaging the example app to see if that causes the issue because webpack won't handle native dependencies in Node and requires something like this in `webpack.config.js`:

```json
    ...
    externals: {
        usb: 'commonjs usb'
    }
    ...
```

strazzere, 2022-07-01:
Making it an `externals` then loading it in the `index.html` prior to the renderer loading will let this work.

closed 2023-02-26 by thegecko