node-usb/node-usb #521: Poll Transfer - Start/Stop Poll

noah-moutfitters, 2022-06-15:
## Bug Description:

Currently not able to make more than one poll transfer without closing the device.

Starting poll transfer has conditional:

```
        if (this.pollTransfers) {
            throw new Error('Polling already active');
        }
```

when no more polls are pending, it resets the state like so:

```
    if (_this.pollPending === 0) {
        _this.pollTransfers = [];
        _this.emit('end');
    }
```

The conditional needs to handle the case of an empty array, since any object in javascript is truthy or the poll transfers need reset to be undefined.  

I can create a PR for this. 

## Steps to Reproduce:
1. Open a usb device as normal
2. Start a poll transfer
3. Close said poll transfer
4. Wait for the end event to emit
5. Start another poll transfer

thegecko, 2022-06-15:
A PR would be great!

thegecko, 2022-06-16:
Does #522 fix you?

noah-moutfitters, 2022-06-16:
@thegecko yes, thanks!

noah-moutfitters, 2022-06-16:
@thegecko Do you also want the `pollActive` flag set to false when `pollsPending` equals 0? Seems like that may be desired behavior, it I’m unsure.

thegecko, 2022-06-16:
Not sure, code was like this before I arrived ;)

thegecko, 2022-06-16:
> @thegecko Do you also want the pollActive flag set to false when pollsPending equals 0? Seems like that may be desired behavior, but I’m unsure.

I added your suggestion, would you mind checking again?

noah-moutfitters, 2022-06-20:
@thegecko Yes, thank you!

closed 2022-06-20 by noah-moutfitters