node-usb/node-usb #523: Rebuilding on ARM not possible

LRYMND, 2022-07-04:
When trying to build an electron app on a Raspberry Pi running the latest RaspberryPi OS, I keep running into the same error.

```
> electron-builder build --linux AppImage --armv7l -c.extraMetadata.main=build/electron/main.js --publish never

  • electron-builder  version=22.14.13 os=5.15.32-v8+
  • loaded configuration  file=package.json ("build" field)
  • public/electron.js not found. Please see https://medium.com/@kitze/%EF%B8%8F-from-react-to-an-electron-app-ready-for-production-a0468ecb1da3
  • loaded parent configuration  preset=react-cra
  • description is missed in the package.json  appPackageFile=/home/volvo/Development/rtvi/package.json
  • electron-rebuild not required if you use electron-builder, please consider to remove excess dependency from devDependencies

To ensure your native dependencies are always matched electron version, simply add script `"postinstall": "electron-builder install-app-deps" to your `package.json`
  • writing effective config  file=dist/builder-effective-config.yaml
  • rebuilding native dependencies  dependencies=usb@1.7.2 platform=linux arch=armv7l
  • install prebuilt binary  name=usb version=1.7.2 platform=linux arch=armv7l napi=
  • build native dependency from sources  name=usb
                                          version=1.7.2
                                          platform=linux
                                          arch=armv7l
                                          napi=
reason=prebuild-install failed with error (run with env DEBUG=electron-builder to get more information)
                                          error=prebuild-install info begin Prebuild-install version 5.3.6
    prebuild-install WARN install prebuilt binaries enforced with --force!
    prebuild-install WARN install prebuilt binaries may be out of date!
    prebuild-install info looking for cached prebuild @ /home/volvo/.npm/_prebuilds/5d82ba-usb-v1.7.2-napi-v4-linux-armv7l.tar.gz
    prebuild-install http request GET https://github.com/tessel/node-usb/releases/download/v1.7.2/usb-v1.7.2-napi-v4-linux-armv7l.tar.gz
    prebuild-install http 404 https://github.com/tessel/node-usb/releases/download/v1.7.2/usb-v1.7.2-napi-v4-linux-armv7l.tar.gz
    prebuild-install WARN install No prebuilt binaries found (target=4 runtime=napi arch=armv7l libc= platform=linux)

```    

Tried building on a Raspberry 3 and 4 with clean installations of the latest OS.

Node v16.15.1
NPM v8.11.0.
Electron v13.6.9

Libudev is installed.

LRYMND, 2022-07-07:
It was a compatibility issue with Bullseye. Downgraded the OS to buster and now its working.

closed 2022-07-07 by LRYMND

thegecko, 2022-07-08:
> It was a compatibility issue with Bullseye. Downgraded the OS to buster and now its working.

Thanks for the details. Moving forwards we probably want to be able to support bullseye, so anything we need to be aware of?

LRYMND, 2022-07-17:
In the end I think that the build actually succeeded since it created an AppImage in the build directory. However the AppImage wouldn't execute. I could imagine that there was an issue with permissions regarding node-usb/libusb although I updated the udev rules for the specific device. Didn't investigate further though and just downgraded the OS.

gniezen, 2022-12-07:
I just ran into the same issue with Raspberry Pi OS Bullseye. It built just fine, but when you launch the app you get the following error:

```
Uncaught error: No native build was found for platform=linux arch=arm runtime=electron abi=106 uv=1 armv=default libc=glibc node=16.14.2 electron=19.0.8
```