node-usb/node-usb #529: Initial detaching of a device doesn't trigger detach event.

Newbrict, 2022-08-12:
## Bug Description:

Starting the app with the device plugged in, then physically unplugging it doesn't trigger `usb.on('detach', (device) => {`. If I unplug, then plug, then unplug the device again, then my detach event finally fires. Also, if I start the app with the device unplugged, it works on the first detach. 

It seems like a detach event can't happen before an attach event, or something. I noticed that a bunch of attach events are fired when *detaching* initially. But none have my device id, so I don't know what's going on. Please advise. Thank you.

## Steps to Reproduce:
1. Plug in your device
2. Detach it (only attach events are fired)
3. Plug it back in (attach event is fired with that device id)
4. Detach it again, this time you will see that device id being detached.

## Additional Information
- Operating System: Windows 10 64 bit.
- Runtime node v16.16.0
- Node USB Version (`npm ls usb`): 2.5.0

mildsunrise, 2022-08-12:
does it happen in the version just before (v2.4.3)? we've just touched hotplug detection on windows (#524)

Newbrict, 2022-08-12:
> do both these things (the initial detach events, and the absence of a detach event for your device) happen in the version just before (v2.4.3)? we've just touched hotplug detection on windows (#524)

I will figure out how to downgrade the package, test that, and report back. Eta 30 minutes.

Newbrict, 2022-08-12:
So the behavior on `v2.4.0` seems completely broken. It continuously fires attach and detach events every second or so, even though I'm not touching any USB devices. v2.5.0 did at least only respond to devices being plugged or unplugged, albeit not completely accurately.

Newbrict, 2022-08-12:
I verified this with another device, same behavior. I am not familiar with this code to dig into the issue, but it seems like it could be caused by not having an accurate initial representation of what's plugged in and what's not.

Newbrict, 2022-08-12:
> So the behavior on `v2.4.0` seems completely broken. It continuously fires attach and detach events every second or so, even though I'm not touching any USB devices. `v2.5.0` did at least only respond to devices being plugged or unplugged, albeit not completely accurately.

Ah, I tested with the version you specified `v2.4.3` and it worked properly there. I will be using that version until this issue is figured out -- thanks.

thegecko, 2022-08-13:
OK, so this is only an issue on Windows and I think it's due to the list of initial devices not being set before the attach/detach listeners being attached.

I can see the problem, but am on holiday and don't have Windows access to test it. I can push a blind fix to a branch for others to try, though if we want to crowdsource a quick fix :)

Newbrict, 2022-08-15:
Sorry for the delay, I've been working on something else, but I will get back to this ASAP.

closed 2022-08-29 by thegecko

thegecko, 2022-08-29:
Fixed in v2.5.1

PR #531 (2022-08-29): Fix initial windows device list
## Changes

- Set initial device list for windows polling

## Fixes

- Fixes #529

## Checklist

Have you...

- [x] Tested the change acts as expected
- [x] Checked the change doesn't remove or change existing functionality
- [x] Considered how the feature impacts the product and tested around it (not just the happy paths)
- [x] Where possible, executed the hardware tests on multiple operating systems

cc @mildsunrise  @Newbrict @Julusian