node-usb/node-usb #530: Can the build NOT use yarn?

gre, 2022-08-12:
### Feature Description:

At the moment, you are documenting the library prerequisite to be `[Node.js >= v10.20.0](https://nodejs.org/), which includes npm.`

Unfortunately it appears that you also require the user-land to also have `yarn` installed. Which isn't the case for us (at Ledger) as we moved to `pnpm` now.

During the prebuild phase, we're blocked by

```
ledger-live-desktop:build:     npm ERR! 'yarn' is not recognized as an internal or external command,
```

Would it be possible to make the build not using yarn but just `npm run {command}` ?

Thanks,
gre

thegecko, 2022-08-12:
Yarn is used to build this project,  but the native binaries are then shipped with the release.  You shouldn't need to rebuild node-usb.

gre, 2022-08-12:
but it what seems to happen when we depends on this library 🤔  (possibly when prebuild aren't available for us, typically we likely install & rebuild for electron)

thegecko, 2022-08-13:
@gre Please confirm if this fixed you.

mildsunrise, 2022-08-13:
not sure what's going on here (not very familiar with electron), but users shouldn't need yarn even if no prebuild is available. at least I didn't need it. if this has changed we should fix it

thegecko, 2022-08-13:
It's due to the dependent scripts, e.g.:

https://github.com/node-usb/node-usb/blob/master/package.json#L39

mildsunrise, 2022-08-13:
hmm but the prepare script should only run when you do `npm install` directly on node-usb source. it should never run if `npm install` finds `usb` as a dependency (unless you specify a git repo). not sure why Electron would need to run `prepare` :thinking:

thegecko, 2022-08-13:
I believe electron-rebuild runs these scripts, which used to be required prior to NAPI support. TBF, I can probably refactor those scripts anyway and remove the yarn bits.

thegecko, 2022-08-13:
It looks like `prepare` may be called when packages consuming this module are packaged?

https://docs.npmjs.com/cli/v8/using-npm/scripts#life-cycle-scripts

mildsunrise, 2022-08-16:
I would be surprised if it did, since dependencies aren't included when packaging. are you referring to a particular point in the document you linked?

thegecko, 2022-08-16:
Yeah,  there's a note in the prepare section

mildsunrise, 2022-08-16:
> NOTE: If a package being installed through git contains a prepare script, its dependencies and devDependencies will be installed, and the prepare script will be run, before the package is packaged and installed

"the prepare script will be run" means only of the package being installed through git, not the dependencies themselves AFAIK. this only applies if somone specifies `"usb": "<git URL>"` in their package.json

thegecko, 2022-08-16:
Ok

closed 2022-08-29 by thegecko

thegecko, 2022-08-29:
Fixed in v2.5.1