node-usb/node-usb #532: Node-usb fails to build in node runtime when `DOM` lib isn't included

gcampbell-msft, 2022-08-15:
## Bug Description:
When attempting to use this package in a typescript package that is to be used in a node runtime, as well as defining that the only `lib` in the `tsconfig.json` should be `es2019`, the project fails to build.

This is because the typescript compiler seems to include the `DOM` library by default, but when it is specified that it shouldn't be included because this is a node project, not a browser context, it fails to build. 

## Steps to Reproduce:
1. Unzip the attached zip file.
[webUsbIssue.zip](https://github.com/node-usb/node-usb/files/9338009/webUsbIssue.zip)

2. Attempt to build the project by running `tsc -p ./

Expected: Successful build
Actual: Failed build due to `dom` libraries not being available. 

## Additional Information
- Operating System: Windows 11
- Runtime (`node -v` or `electron -v`): 16.13.1
- Node USB Version (`npm ls usb`): 2.5.0

mildsunrise, 2022-08-15:
dupe of / see https://github.com/node-usb/node-usb/issues/520#issuecomment-1145280361

this is expected since we implement WebUSB (a browser API) and the types depend on some things (such as EventTarget) that have only recently been made available in Node.js

thegecko, 2022-08-29:
We switched to using the WebUSB by default, so this is expected.

closed 2022-08-29 by thegecko

gcampbell-msft, 2022-08-29:
@thegecko Does this mean that even in node contexts, the node-usb package will require including the DOM libraries?

thegecko, 2022-08-29:
Yes, the WebUSB API uses DOM events

gcampbell-msft, 2022-08-29:
@thegecko I believe you're correct that it's only the types. 

I think a typescript example could be useful. 

I think that creating a node project and wanting to only include the libraries necessary (and not the `DOM` libraries) is likely a common scenario. I do wonder if the packages should be broken up such that a user wouldn't have to pull in the `DOM` libraries even though it's a node project. 

Thanks!

thegecko, 2022-08-29:
The project decided to [switch to the WebUSB API](https://github.com/node-usb/node-usb/issues/401) but as this is a web API which uses web events, I can't see how we can avoid using the DOM types unless we explicitly copy out the types we want to use. This approach would then require us to maintain that copy of the types.

The older API which doesn't use WebUSB is being maintained as a legacy approach really. Splitting it out takes us back to the situation before where this library existed and I maintained the [webusb library](https://github.com/thegecko/webusb) on top.

mildsunrise, 2022-08-29:
yes, it's only the types. no effect on compiled JS