node-usb/node-usb #533: Rapid disconnect/reconnect of device isn't recognized.

gcampbell-msft, 2022-08-15:
## Bug Description:
Rapid disconnection and reconnection doesn't seem to be properly recognized. 

This is related to this bug posted here: https://github.com/microsoft/vscode-serial-monitor/issues/26, where someone posted this issue on the serial monitor. 

## Steps to Reproduce:
1. Plug in a device
2. Unplug the device and **quickly** plug the device back in.

Expected: The package correctly picks up both events and the device is properly picked up on reconnection
Actual: It doesn't seem to handle the reconnection event, it seems that reconnections happen too quickly for it to handle? 

## Additional Information
- Operating System: Windows 11
- Runtime (`node -v` or `electron -v`): 16.13.1
- Node USB Version (`npm ls usb`): 2.5.0

mildsunrise, 2022-08-15:
does this happen in the version just before (v2.4.3)? we've just touched hotplug detection on windows (https://github.com/node-usb/node-usb/pull/524)

gcampbell-msft, 2022-08-15:
I just realized I put the wrong version for this bug, this is happening on version `v2.4.3`.

mildsunrise, 2022-08-15:
for context: hotplug detection has never really worked correctly in Windows until 2.5.0. before that, we regularly polled the current devices and relied on libusb to tell us whether two devices are the same or not.

if libusb is still recognizing a device as "the same" after a reconnection, then I'm afraid there's little we can do and this is an issue that libusb should fix.

we could do additional validation by checking that `busNumber` and `deviceAddress` have not changed, but something tells me libusb is already doing that. maybe Windows is reusing the deviceAddress?

in any case, even if it *does* work in 2.5.0, it's essentially a data race and you should not rely on it

mildsunrise, 2022-08-15:
it would be nice to make sure that this is indeed a Windows-only issue, and/or if it happens with a specific device or any device.

I did a quick test on my Linux (hardcoding `hotplugSupported = false` in 2.4.3 and using a very long poll interval) and everything works as expected: Linux assigns a different deviceAddress, and libusb returns a new device

gcampbell-msft, 2022-08-15:
> for context: hotplug detection has never really worked correctly in Windows until 2.5.0. before that, we regularly polled the current devices and relied on libusb to tell us whether two devices are [...]

Does this mean that as far as you know, updating to 2.5.0 should solve the issue (at least as much as it can be solved?)

thegecko, 2022-08-15:
I would expect an update to resolve this issue as I suspect the older polling mechanism on Windows is the culprit.  Easy to test...

gcampbell-msft, 2022-08-15:
> I would expect an update to resolve this issue as I suspect the older polling mechanism on Windows is the culprit. Easy to test...

Can confirm, that it's easy to test, I've been playing with it, and it seems to work. I'll go with this and return to this thread if any issues arise! Thanks!

mildsunrise, 2022-08-15:
this issue has implications beyond hotplug detection... if libusb is returning the same device for a new device (that isn't in the same state) this could lead to bugs in apps that do not use hotplug detection too :thinking: but if it rarely happens I guess we can just ignore it...

thegecko, 2022-08-29:
I think the initial issue is resolved with an update to `node-usb` so closing this for now.

closed 2022-08-29 by thegecko