node-usb/node-usb #534: It is recommended to exclude the source code and map files when publishing the NPM package, and only publish the final product

shibin-tiny-dust, 2022-08-28:
### Feature Description:

mildsunrise, 2022-08-29:
I agree that TypeScript source code should be excluded from the published package, but AFAIK best practice is to leave the compiled JS, d.ts, map files, and the C++ source code, right?

shibin-tiny-dust, 2022-08-29:
I agree with AFAIK. When using typescript, I will focus on its declaration file. D. TS instead of C or C + + source code. The source map file I mentioned is a source map, such as binding.js.map. When I use this library to package production products, I must delete binding.js.map. And C, C + + code` binding. js。 Map ` can only be used when debugging in the development environment, because programmers who only use js or TS may not be able to read C + + or C code. The above is just a little trouble in my personal use. It may not be comprehensive. Thank you for your reply. Because my English is not very good, my expression is not very clear
[image]

thegecko, 2022-08-29:
Current package status is as follows:

## Included

- Native prebuilt binaries
- .vscode config files
- Compiled JavaScript code, typings and map files
- C++ source
- libusb source
- JavaScript source
- test scripts
- config files

## Not included

- API docs (available on the website)

## Proposal

Seeing as the source code is available online, I propose we remove the following

- Typescript and C++ sources from the npm package (in fact the libusb sources we ship are 2.4MB!)
- All config files (root and .vscode)

I think it's good practice to ship the map files and test files to ease testing and remote debugging.

WDYT?

thegecko, 2022-08-29:
Ooh, just realised that removing the source will mean unsupported targets won't be able to node-gyp build the native binaries from source. This may not be an issue if there are only a few of these due to the extensive prebuilds we ship?

mildsunrise, 2022-08-29:
about excluding C/C++ source: we absolutely should not break systems that can't benefit from our prebuilds. even with N-API, there are plenty of reasons this can happen (different libc, such as in embedded systems, older glibc, unsupported arch, etc.). prebuilds are supposed to be an optimization, never an alternative to building from source. (or at least, I don't know of any major package that excludes C++ source. if you do, please post examples)

about excluding the map files: I could agree with that. I'm still skeptical, as end users (and ourselves) may be interested in getting meaningful stack traces in exceptions. if the end user really cares about disk space that much, they can always use a bundler...

about excluding config files, CI, tools, tests, examples: I'm totally fine with that, please go ahead.

> in fact the libusb sources we ship are 2.4MB

that's normal, I'm afraid... good practice is to ship our libusb instead of relying on system installed one. what we could do is trim the sources (if we aren't already doing that) to only the code that is compiled by node-gyp. I do that in [node-bpf](https://github.com/mildsunrise/node_bpf).

Julusian, 2022-08-30:
> about excluding C/C++ source: we absolutely should not break systems that can't benefit from our prebuilds. even with N-API, there are plenty of reasons this can happen (different libc, such as in [...]

I stumbled upon https://github.com/realm/realm-js/issues/4060 a while ago when looking into some native building stuff.
From a quick skim I can't find any real issues caused by the lack of c++ sources over there.

I agree that it would feels bad to break support for weird systems, but is it enough to support the officially supported node platforms? 
If someone is needing to build against an older glibc, then is it likely that they will encounter other potentially blocking issues? such as their g++ being too old for used language features. or could they have issues with libusb not supporting their architecture/os?
This is an area I have been wondering about for other libraries, mostly to figure out what prebuilds are worth producing, but I am now wondering if I should be thinking about omitting their c++ source too..

> if the end user really cares about disk space that much, they can always use a bundler...

While using a bundler isnt that easy to do for targetting nodejs, especially when native modules are involved, I would agree that this would be an important step of reducing on disk size. I have been working on this for one project because I want to reduce the number of files on disk, as windows is terribly slow at working with thousands of tiny files. And in doing so, the result is a 35mb (including a couple of native modules that were excluded from the bundle), compared to the unbundled 110mb

thegecko, 2022-08-31:
I wonder if splitting the package into `node-usb` and `node-usb-src` could be an approach?

Then users with prebuilds just use `node-usb` and more advanced cases could also install `node-usb-src` which is either the full-blown package or just the source used by `node-gyp-build` in `node-usb` along with an optional dependency?

Julusian, 2022-08-31:
I dont think splitting like that is a good solution. I imagine it likely that some of these issues will occur in places where node-usb is a dependency of a dependency. Which will then force users to figure out how to force install `node-usb-src` instead of `node-usb` (which is completely doable via yarn resolutions), and tell it to install a compatible version.
Being a peerDependency is likely to cause pain too, once you consider that different dependencies may require different versions.

I was just wondering if perhaps a step to fetch the source from the github release would work, but that sounds like a bad idea for my next case...

Another use case to consider here, is that some users (ref https://github.com/cmake-js/cmake-js/issues/276) need to be able to produce auditable builds. In this case prebuild-install may intentionally not be allowed to fetch the binaries from github, so it must be built from source with whatever was in the npm package.

thegecko, 2022-08-31:
Could we use the system libusb when in the wild and just remove that source? But continue to build the prebuilds with the submodule?

mildsunrise, 2022-09-01:
> I dont think splitting like that is a good solution. I imagine it likely that some of these issues will occur in places where node-usb is a dependency of a dependency

My thoughts exactly.

> Another use case to consider here, is that some users (ref https://github.com/cmake-js/cmake-js/issues/276) need to be able to produce auditable builds. In this case prebuild-install may [...]

Don't we ship all the prebuilds in the NPM package rather than grabbing from the Internet?

thegecko, 2022-09-01:
> Don't we ship all the prebuilds in the NPM package rather than grabbing from the Internet?

We do

thegecko, 2022-09-09:
Shall we agree to leave the distribution package as-is, then?

PR #536 (2022-09-24): Trim npm package
## Changes

- Removes config files from npm package
- Removes typescript source from npm package

While I'm keen to strip the package down and remove the unneeded config files, removal of the c++ source will mean the native module will no longer build/rebuild on unsupported systems and we would rely entirely on the shipped prebuild binaries.

## Fixes

- Fixes #534 

## Checklist

Have you...

- [x] Tested the change acts as expected
- [x] Checked the change doesn't remove or change existing functionality
- [x] Considered how the feature impacts the product and tested around i [... cut]
closed 2022-09-24 by thegecko